openwrt/luci #3387: A bug of DynamicList

0x112, 2019-12-03:
Cannot select multiple identical values ​​from a list.
That is cannot add:
```luci
list main_server '111'
list main_server '111'
list main_server '111'
list main_server '111'
```

Related issue: https://github.com/shadowsocks/luci-app-shadowsocks/issues/234

jow-, 2019-12-03:
What is the point of having multiple identical entries? The `DynamicList` widget prevents this by design, and I'd prefer to keep it this way.

0x112, 2019-12-03:
@jow- Thank you for your prompt reply.
In https://github.com/shadowsocks/luci-app-shadowsocks project function, it can run multiple processes in a multi-core cpu to load balance, so I hope that DynamicList can add a usage to support it.

jow-, 2019-12-03:
But why does it require the same entries multiple times? Wouldn't it be better to use one select per cpu?

0x112, 2019-12-03:
One line entry represents an instance, it was designed by @aa65535

0x112, 2019-12-03:
Also you can try: https://github.com/shadowsocks/luci-app-shadowsocks/releases/tag/v1.9.1, provide English language too.

ysc3839, 2019-12-03:
@0x112 I would suggest you using [luci-app-shadowsocks-libev](https://github.com/openwrt/luci/tree/master/applications/luci-app-shadowsocks-libev), which is one instance per uci section.
[image]
```
config server 'svr'
	option method 'aes-128-gcm'
	option server_port '2333'
	option server '127.0.0.1'

config ss_redir 'red1'
	option mode 'tcp_and_udp'
	option local_port '4666'
	option server 'svr'

config ss_redir 'red2'
	option mode 'tcp_and_udp'
	option local_port '4666'
	option server 'svr'
```

0x112, 2019-12-04:
@ysc3839 But I think the original version of shadowsocks luci may handle the details better, I prefer to use the original.

0x112, 2019-12-04:
@jow- @ysc3839 Why DynamicList doesn't use openwrt-18.06 version's logic? and openwrt-18.06 version can allows items to reselect values.

jow-, 2019-12-04:
> Why DynamicList doesn't use openwrt-18.06 version's logic?

Because parts of it has been rewritten. I don't like the duplicate values thing at all and have no plans to make this the default behavior. Maybe I'm going to add an `allowduplicates` flag, but not sure when I'll get around to it.

0x112, 2019-12-04:
Thanks, hope to be released before the release version of openwrt-19.07. :smile:

closed 2025-02-24 by systemcrash