openwrt/luci #3389: uci get or config_get get the wrong flag

rufengsuixing, 2019-12-04:
https://user-images.githubusercontent.com/26851715/69961402-beba5d00-1546-11ea-9f35-3f20c76ed3e5.png
the result of
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled)
test=$(uci get AdGuardHome.@AdGuardHome[0].enabled 2>/dev/null)
is different
https://github.com/rufengsuixing/luci-app-adguardhome/issues/7

jow-, 2019-12-04:
This application is not part of the LuCI repository.

closed 2019-12-04 by jow-

rufengsuixing, 2019-12-04:
not the application，just the uci shell command not work

jow-, 2019-12-04:
The uci shell is neither. However I have never seen such a bug and from a simple test I cannot reproduce it either. Make sure that nothing changed the config between the get calls. Maybe use `uci changes AdGuardHome` to double check.