openwrt/luci #3395: luci-app-ddns displays error on load

shirogitsune, 2019-12-09:
Upgrading OpernWRT on my TP-Link 1043ND v1 and completely clearing the configurations due to some weird behavior and instability.

I Installed `luci-app-ddns` version 2.4.9-3 and when navigating to the "Services" > "Dynamic DNS" menu item, I am confronted with the following error message:
```
Failed to execute cbi dispatcher target for entry '/admin/services/ddns'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/ddns.lua:107: attempt to index field '?' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/ddns.lua:107: in function 'service_version'
	/usr/lib/lua/luci/controller/ddns.lua:117: in function 'service_ok'
	/usr/lib/lua/luci/model/cbi/ddns/overview.lua:20: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:845: in function </usr/lib/lua/luci/dispatcher.lua:839>
```
The `ddns-scripts` package is version 2.7.8-1 and the configuration file is the default one generated upon install.

**Release Info:**
```
root@OpenWrt:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="18.06.5"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 18.06.5"
VERSION_ID="18.06.5"
HOME_URL="http://openwrt.org/"
BUG_URL="http://bugs.openwrt.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r7897-9d401013fc"
LEDE_BOARD="ar71xx/generic"
LEDE_ARCH="mips_24kc"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://openwrt.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt 18.06.5 r7897-9d401013fc"
```

**Installed Packages:**
```
root@OpenWrt:~# opkg list-installed
base-files - 194.2-r7897-9d401013fc
busybox - 1.28.4-3
ddns-scripts - 2.7.8-1
dnsmasq - 2.80-1.4
dropbear - 2017.75-7.1
firewall - 2018-08-13-1c4d5bcd-1
fstools - 2019-03-28-ff1ded63-5
fwtool - 1
hostapd-common - 2018-05-21-62566bc2-8
ip6tables - 1.6.2-3
iptables - 1.6.2-3
iw - 4.14-1
iwinfo - 2018-07-31-65b8333f-1
jshn - 2018-07-25-c83a84af-2
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 4.9.198-1-2b8f9dfe583e5c09aadd9474da55137f
kmod-ath - 4.9.198+2017-11-01-10
kmod-ath9k - 4.9.198+2017-11-01-10
kmod-ath9k-common - 4.9.198+2017-11-01-10
kmod-cfg80211 - 4.9.198+2017-11-01-10
kmod-gpio-button-hotplug - 4.9.198-2
kmod-ip6tables - 4.9.198-1
kmod-ipt-conntrack - 4.9.198-1
kmod-ipt-core - 4.9.198-1
kmod-ipt-nat - 4.9.198-1
kmod-lib-crc-ccitt - 4.9.198-1
kmod-mac80211 - 4.9.198+2017-11-01-10
kmod-nf-conntrack - 4.9.198-1
kmod-nf-conntrack6 - 4.9.198-1
kmod-nf-ipt - 4.9.198-1
kmod-nf-ipt6 - 4.9.198-1
kmod-nf-nat - 4.9.198-1
kmod-nf-reject - 4.9.198-1
kmod-nf-reject6 - 4.9.198-1
kmod-nls-base - 4.9.198-1
kmod-ppp - 4.9.198-1
kmod-pppoe - 4.9.198-1
kmod-pppox - 4.9.198-1
kmod-slhc - 4.9.198-1
kmod-usb-core - 4.9.198-1
kmod-usb-ehci - 4.9.198-1
kmod-usb-ledtrig-usbport - 4.9.198-1
kmod-usb2 - 4.9.198-1
libblobmsg-json - 2018-07-25-c83a84af-2
libc - 1.1.19-2
libgcc - 7.3.0-2
libip4tc - 1.6.2-3
libip6tc - 1.6.2-3
libiwinfo - 2018-07-31-65b8333f-1
libiwinfo-lua - 2018-07-31-65b8333f-1
libjson-c - 0.12.1-2
libjson-script - 2018-07-25-c83a84af-2
liblua - 5.1.5-1
liblucihttp - 2019-07-05-a34a17d5-1
liblucihttp-lua - 2019-07-05-a34a17d5-1
libmbedtls - 2.16.3-1
libnl-tiny - 0.1-5
libpthread - 1.1.19-2
libubox - 2018-07-25-c83a84af-2
libubus - 2018-10-06-221ce7e7-1
libubus-lua - 2018-10-06-221ce7e7-1
libuci - 2019-05-17-f199b961-1
libuclient - 2018-11-24-3ba74ebc-1
libustream-mbedtls - 2018-07-30-23a3f283-2
libxtables - 1.6.2-3
logd - 2018-02-14-128bc35f-2
lua - 5.1.5-1
luci - git-19.309.48729-bc17ef6-1
luci-app-ddns - 2.4.9-3
luci-app-firewall - git-19.309.48729-bc17ef6-1
luci-base - git-19.309.48729-bc17ef6-1
luci-lib-ip - git-19.309.48729-bc17ef6-1
luci-lib-jsonc - git-19.309.48729-bc17ef6-1
luci-lib-nixio - git-19.309.48729-bc17ef6-1
luci-mod-admin-full - git-19.309.48729-bc17ef6-1
luci-proto-ipv6 - git-19.309.48729-bc17ef6-1
luci-proto-ppp - git-19.309.48729-bc17ef6-1
luci-ssl - git-19.334.34552-3a3d8f4-1
luci-theme-bootstrap - git-19.309.48729-bc17ef6-1
mtd - 23
netifd - 2019-01-31-a2aba5c7-2.1
odhcp6c - 2018-07-14-67ae6a71-15
odhcpd-ipv6only - 1.15-3
openwrt-keyring - 2018-05-18-103a32e9-1
opkg - 2019-06-14-dcbc142e-1
ppp - 2.4.7-12
ppp-mod-pppoe - 2.4.7-12
procd - 2018-03-28-dfb68f85-1
px5g-mbedtls - 6
rpcd - 2018-11-28-3aa81d0d-2
rpcd-mod-rrdns - 20170710
swconfig - 11
uboot-envtools - 2018.03-1
ubox - 2018-02-14-128bc35f-2
ubus - 2018-10-06-221ce7e7-1
ubusd - 2018-10-06-221ce7e7-1
uci - 2019-05-17-f199b961-1
uclient-fetch - 2018-11-24-3ba74ebc-1
uhttpd - 2018-11-28-cdfc902a-3
usign - 2019-08-06-5a52b379-1
wireless-regdb - 2017-10-20-4343d359
wpad-mini - 2018-05-21-62566bc2-8
```

**rpcd info:**
```
root@OpenWrt:~# ps ww | grep rpcd
  835 root      1528 S    /sbin/rpcd -s /var/run/ubus.sock -t 30
14684 root      1200 S    grep rpcd
```

**board info:**
```
root@OpenWrt:~# lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))'
board_name      tl-wr1043nd
hostname        OpenWrt
release table: 0x431cb0
        distribution    OpenWrt
        revision        r7897-9d401013fc
        version 18.06.5
        target  ar71xx/generic
        description     OpenWrt 18.06.5 r7897-9d401013fc
model   TP-Link TL-WR1043N/ND v1
kernel  4.9.198
system  Atheros AR9132 rev 2
```

**ubus Login:**
```
root@OpenWrt:~# ubus call session login '{ "username": "root", "password": "********" }'
{
        "ubus_rpc_session": "08c797a0968473e16845b11bbe1b1760",
        "timeout": 300,
        "expires": 299,
        "acls": {
                "access-group": {
                        "uci-access": [
                                "read",
                                "write"
                        ],
                        "unauthenticated": [
                                "read"
                        ]
                },
                "ubus": {
                        "session": [
                                "access",
                                "login"
                        ]
                },
                "uci": {
                        "*": [
                                "read",
                                "write"
                        ]
                }
        },
        "data": {
                "username": "root"
        }
}
```

feckert, 2019-12-09:
@Ansuel any opinion? I think the root cause of the problem is this line https://github.com/openwrt/luci/blob/openwrt-18.06/applications/luci-app-ddns/luasrc/controller/ddns.lua#L117
@shirogitsune Could you change this line and return always true?

Ansuel, 2019-12-09:
problem is that for some reason

`ver = IPKG.info(srv_name)[srv_name].Version`

IPKG.info can't find srv_name (ddns_scripts) and fail to access the key with index srv_name
A bug in IPKG.info ? Or the module was different in 18.06 ?

jow-, 2019-12-09:
The `module.ipkg` class is extremely expensive, isn't there some more lightweight way to obtain the installed version?

Ansuel, 2019-12-09:
Well i think use the os.popen is equally expensive... Should we use it and remove ipkg module ?

shirogitsune, 2019-12-09:
> @Ansuel any opinion? I think the root cause of the problem is this line https://github.com/openwrt/luci/blob/openwrt-18.06/applications/luci-app-ddns/luasrc/controller/ddns.lua#L117
> @shirogitsune Could you change this line and return always true?

@feckert. I tried that along with verifying that the helper script returned a valid version number.

Ansuel, 2019-12-09:
you need to clear luci cache

rm -r /tmp/luci-*

shirogitsune, 2019-12-09:
@Ansuel I was going to try what you suggested once I got home from work. However, I clicked on the "Dynamic DNS" and it has apparently decided to begin working. I checked the lua script and it was unmodified so I cleared the luci cache as you suggested and it's still working. One of three things has apparently happened:
1. You used some sort of quantum mechanics to realign the bits so it is working on my router.
2. I have gone completely mad. (Not ruling this one out).
3. The an extended power outage bled the electrons out of RAM and unstuck whatever was broken.

I suspect #3 is what fixed it, but I'm not going to rule out 1 or 2. Thanks everyone.

closed 2019-12-09 by shirogitsune

gdamjan, 2020-05-04:
Slightly related to this, I'm getting the following error, and I suspect it is because I don't have opkg installed at all (I build my own image with all I need). Should opkg be optional somehow, or at least specified as a package dependency?
```
Failed to execute cbi dispatcher target for entry '/admin/services/ddns'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/ddns.lua:116: attempt to index field '?' (a nil value)
stack traceback:
	/usr/lib/lua/luci/controller/ddns.lua:116: in function 'service_version'
	/usr/lib/lua/luci/controller/ddns.lua:126: in function 'service_ok'
	/usr/lib/lua/luci/model/cbi/ddns/overview.lua:20: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:943: in function </usr/lib/lua/luci/dispatcher.lua:937>
```
(this is on 19.07-HEAD)

qingshengwen, 2020-11-02:
> you need to clear luci cache
> 
> rm -r /tmp/luci-*

yes, It works