openwrt/luci #3398: Changing the LAN IP from LuCI failed in openwrt-19.07 when change this via LAN IP

srdgame, 2019-12-10:
Our device is using openwrt-19.07 branch.

We changed our device default LAN IP to 192.168.1.248 and disabled DHCP on LAN.

By accessing the LuCI via http://192.168.1.248, then changes the LAN ip to 192.168.1.249.  The box's LAN IP changes to 192.168.1.249, but the LuCI confirmation post still using 192.168.1.248 as the box's IP. 

Thus the changes will be rollback, and we failed to change the LAN IP.

feckert, 2019-12-10:
In master there is a new option on the button Apply&Save on the Network page to apply without verify the configuration.
**APPLY UNCHECKE**D
I hope this will solve the issue with your LAN-IP

jow-, 2019-12-10:
It is impossible to solve it as the new IP is not known in advance. Use the Apply unchcked mode or the Apply unchecked button appearing after the failed rollback.

closed 2019-12-10 by jow-

srdgame, 2019-12-10:
@feckert @jow- Thanks