openwrt/luci #3399: add boot_service function for init.d

rufengsuixing, 2019-12-10:
```
boot() {
		procd_open_instance "waitnet"
		procd_set_param command "/usr/share/AdGuardHome/waitnet.sh"
		procd_set_param respawn ${respawn_threshold:-3600} ${respawn_timeout:-1} ${respawn_retry:-1}
		procd_close_instance
}
```
if i write like this ,init won`t start it
so we need to add a function in rc.common for procd which will call boot_service
```
boot() {
		rc_procd boot_service "$@"
		if eval "type service_started" 2>/dev/null >/dev/null; then
			service_started
		fi
}
```

jow-, 2019-12-10:
Procd is not maintained in this repository. I am not sure what you want to achieve with the script above but it seems to me like you want to block init? Maybe consider using hotplug scripts instead - they're better suited to react on network events without having to block the init process.

closed 2019-12-10 by jow-

rufengsuixing, 2019-12-10:
in order not to block init,i use procd_open_instance.if i put the sh in the boot function ,it will block the init for sure