openwrt/luci #3405: luci-mod-network: add support for multiple dnsmasq sections

treysis, 2019-12-11:
In `/etc/config/dhcp` you can configure multiple `config dnsmasq` sections in cases where you need different configurations for dnsmasq on different interfaces/vlans, e.g. `config dnsmasq 'dns_vlan1'` and `config dnsmasq 'dns_vlan2'`. Currently, luci-mod-network will only show the first `config dnsmasq` section. This should probably be implemented as several tabs, one for each section.

y-alexey, 2020-01-12:
> In `/etc/config/dhcp` you can configure
Do you try it's working? I cant make it work some time. Only one dnsmasq instance started for me.

treysis, 2020-01-12:
Yes, it is working. It should be configured something like this:
```
config dnsmasq 'dns_1'
    ...
config dnsmasq 'dns_2'
    ...
```
The other sections (config dhcp, config odhcpd, ...) can't have the same name for some reason that I can't explain anymore.

feckert, 2020-01-13:
Regardless of the type of a section in the uci. The name of a section can only be used once. This is a restriction of uci.

treysis, 2020-08-09:
Any idea how to solve this?

treysis, 2020-08-11:
Ok I just noticed this has been there for quite some time in some preliminary form...you just have to scroll down on the configuration page 😅 However, this only works for already created sections. It's not possible to create multiple instances from LuCi, so this still needs some work.

thekad, 2022-05-05:
The above should be said many times, I was also unaware that you could control many instances of dnsmasq with stock openwrt (although with some effort) thank you so much @treysis for the tip :)
The current use case is (probably not *that* uncommon) having 2 networks one unrestricted and one restricted with e.g. opendns. If you set the DNS servers per interface it _appears_ to work however all of the servers end up in the same file `/tmp/resolv.conf.auto` and used by the single instance of dnsmasq. This means your DNS queries will be randomly answered by restricted/unrestricted DNS Servers. The solution is to spawn multiple dnsmasq and point them to different (static) resolv.conf files, which this little piece of buried info makes possible.

joaquinvacas, 2023-07-24:
Has this improved by any means?

systemcrash, 2023-09-19:
A naïve implementation can be made by changing perhaps 2-3 lines of luci code in `dhcp.js` today. I have not considered deeper repercussions, but perhaps a rewrite of some GUI elements is in order to make things clearer. 

Those two lines make it look like (attempting to use non-anonymous section naming):

[image]

[image]

This gives access to trivial add/delete of /etc/config/dhcp:dnsmasq sections.

A tabbed presentation might seem more tidy and accessible, but requires a deeper refactoring of how the existing tabs are already presented.

Neustradamus, 2024-08-30:
Have you progressed on this feature?

systemcrash, 2024-08-30:
It’s in LuCI since a while back.

closed 2024-08-30 by systemcrash