openwrt/luci #3408: luci-app-firewall: shouldn't it be the other way around?

castillofrancodamian, 2019-12-13:
[image]
[image]

jow-, 2019-12-13:
Not sure what you mean but there is nothing reversed here.

Source zone should be your upstream, most comonly wan. External port is the port exposed to the upstream network. Destination zone is the network you're forwarding the port to and internal port is the port of the service you're forwarding to.

This also aligns with what the text description is showing. To me it looks as if you simply swapped lan and wan zones when you declared that port forward.

closed 2019-12-13 by jow-

castillofrancodamian, 2019-12-13:
You are right. I was upside down. Well, actually Firewall areas.