openwrt/luci #3426: Issue with theme since "render menu on client side"

solidus1983, 2019-12-22:
I have noticed since 5 days ago there has been issues on the theme i am working on as well as glitches inside pages such as Routing.

Now the theme has been working fine for a while and i didn't make any changes to the files ran a new compile run and was hit with the following error :

```
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'opkg'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/opentomato/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/opentomato/he..."]:20: attempt to call field 'node_childs' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/opentomato/he..."]:20: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:683: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:683: in function 'include'
	[string "/usr/lib/lua/luci/view/opkg.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:913: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:440: in function </usr/lib/lua/luci/dispatcher.lua:439>
```

So i looked at LUCI git and noticed that luci-theme-material has had some changes to the header.htm, footer.htm and the script.js file so as the theme is based off luci-theme-material i simply copied them files over and the theme sort of works however icons have gone missing that was on the Status etc sections but there is also some issues with Status -> Routes which even shows up on luci-theme-material where there is either %gretap or %sip showing in the target column.

Just thought i'd let you know about this just incase its a bug.

Also Rosy theme also comes out with the same error

```
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'view'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/rosy/header'.
A runtime error occurred: [string "/usr/lib/lua/luci/view/themes/rosy/header.h..."]:20: attempt to call field 'node_childs' (a nil value)
stack traceback:
	[string "/usr/lib/lua/luci/view/themes/rosy/header.h..."]:20: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:683: in function 'include'
	[string "/usr/lib/lua/luci/view/header.htm"]:3: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:683: in function 'include'
	[string "/usr/lib/lua/luci/view/view.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85>
	(tail call): ?
	/usr/lib/lua/luci/dispatcher.lua:846: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:440: in function </usr/lib/lua/luci/dispatcher.lua:439>
```

panda-mute, 2019-12-22:
Luci has change to client render the menu, so you need to update your theme.
For your reference:
https://github.com/openwrt-develop/luci-theme-atmaterial/commit/6f8a51ae4d0e43807ef501d736a998d2f0a5dcfc

panda-mute, 2019-12-22:
css change you only need focus on menu data title change:
.main > .main-left > .nav > .slide > a[data-title="status"]:before {
    content: "\e026";
    content: "\e026";
}
}

panda-mute, 2019-12-22:
This seems a bug:
there is also some issues with Status -> Routes which even shows up on luci-theme-material where there is either %gretap or %sip showing in the target column.

solidus1983, 2019-12-22:
Thanks for the reply bud. Will get the fixes in place, then let you know.

solidus1983, 2019-12-22:
Noticed another issue the Realtime Graphs and Graphs area is missing its tabs and it does this too on material theme too.

closed 2020-01-04 by solidus1983