openwrt/luci #3430: [logs] content rendered illegible

n8v8R, 2019-12-23:
- Master 4.19.88 armv7l (mvebu)
- libuci/base/2019-12-12
- libuci-lua/base/2019-12-12
- luci-lib-jsonc/luci/git-19.356.80423
- luci-base/luci/git-19.356.80423
- luci/luci/git-19.356.80423
- luci-i18n-base-en/luci/git-19.356.80423
- luci-compat/luci/git-19.356.80423
- syslog-ng 3.25.1
___
tried different themes | browsers but all met with  same illegible render however (log files are legible via ssh)

[image]

[image]

dibdot, 2019-12-23:
Is this a turris omnia device? If so, not reproducible with latest stable (TOS 4.03) ...
[image]

n8v8R, 2019-12-23:
I am aware that it does not reproduce with OpenWrt 18.06.x (TOS4.x) or 19.07 (TOS5.x) and until 2 days ago it did not exhibit the issue on Master (TOS6.x) either. But latest changes in the aforementioned LuCI packages seems to have introduced the bug.

dibdot, 2019-12-23:
Again, which device and what OS are you using? Plain vanilla OpenWrt or a TOS variant?

n8v8R, 2019-12-23:
TOS, however the issue is not caused by the downstream repo since LuCI packages are not being patched.

jow-, 2019-12-23:
Not reproducible on OpenWrt master. Maybe something is wrong with `cgi-io` on "TOS" ?

n8v8R, 2019-12-23:
It might be something related to file permissions, considering this been caught by the browser tool box

[image]
___
closing since it does not reproduce on OpenWrt, taking it up with the downstream distro instead

jow-, 2019-12-23:
@n8v8R - do you maybe use `luci-nginx` by any chance? It probably needs to be updated to handle the new `/cgi-bin/cgi-exec` applet as CGI executable. Right now it seems as if it gets downloaded instead.

n8v8R, 2019-12-23:
@jow- No, neither that LuCI applet nor nginx at all. 

TOS leverages:
* lighttpd 1.4.54-1.2
* lighttpd-mod-cgi 1.4.54-1.2
* lighttpd-mod-fastcgi 1.4.54-1.2
* cgi-io 16.0
* python3-cgi 3.8.1-3.8-1.0
____
>Right now it seems as if it gets downloaded instead.

Indeed, a seems to be a content type (text/plain) issue somehow. Could this be caused by the python3 module?

n8v8R, 2019-12-23:
`stat /www/cgi-bin/cgi-exec`

>   File: /www/cgi-bin/cgi-exec -> ../../usr/libexec/cgi-io
>   Size: 24              Blocks: 8          IO Block: 4096   symbolic link
> Device: 10h/16d Inode: 57911       Links: 1
> Access: (0777/lrwxrwxrwx)  Uid: (    0/    root)   Gid: (    0/    root)
> Access: 2019-12-21 07:15:24.223114550 +0000
> Modify: 2019-12-19 15:26:42.360646998 +0000
> Change: 2019-12-19 15:27:18.723061137 +0000
>  Birth: -

`stat /usr/libexec/cgi-io`

>   File: /usr/libexec/cgi-io
>   Size: 16498           Blocks: 40         IO Block: 4096   regular file
> Device: 10h/16d Inode: 57906       Links: 1
> Access: (0755/-rwxr-xr-x)  Uid: (    0/    root)   Gid: (    0/    root)
> Access: 2019-12-21 07:15:11.092251910 +0000
> Modify: 2019-12-19 06:51:33.000000000 +0000
> Change: 2019-12-19 15:27:18.753063129 +0000
>  Birth: -

n8v8R, 2019-12-24:
@jow- Narrowed it down to `luci-mod-status`, upgrading it from git-19.352.74009 to git-19.356.80423

Downstream is not patching this package in their build root and thus this seems to be an issue originating from the OpenWrt code source instead.

During the upgrade it exhibits:

> Removing obsolete file /usr/lib/lua/luci/controller/admin/status.lua.
> Removing obsolete file /usr/lib/lua/luci/view/admin_status/dmesg.htm.
> Removing obsolete file /usr/lib/lua/luci/view/admin_status/iptables.htm.
> Removing obsolete file /usr/lib/lua/luci/view/admin_status/syslog.htm.

That correlates with the log rendering issues reported and the observation that the Firewall status is neither working but instead showing a spinner alongside _Collecting data..._

jow-, 2019-12-24:
You‘re using LuCI on lighttpd abd from rhe symptoms you describe, it‘s CGI setup is broken. It seems as if `/cgi-bin/cgi-exec` is not executed, but served as file by your webserver. Both the use of lighttpd and the broken CGI configs are downstream deviations from OpenWrt master defaults.

closed 2019-12-24 by jow-

n8v8R, 2019-12-24:
@jow- Had a closer look in the browser console and *lighthttpd* furnishes correctly

>Content-Type: application/octet-stream

However, https://github.com/openwrt/luci/commit/939b371bc72f4c020c499c815f636dafc296829a introduced 

`+ "/sbin/logread -e ^": [ "exec" ],`

which is not leveraged downstream, but *syslog-ng* and thus the logs are not being read.
___
As for the firewall status spinner alongside Collecting data.. I am not sure but noticed

` /usr/sbin/ip6tables --line-numbers -w -nvxL -t nat` producing

> ip6tables v1.8.3 (legacy): can't initialize ip6tables table `nat': Table does not exist (do you need to insmod?)
> Perhaps ip6tables or your kernel needs to be upgraded.

n8v8R, 2020-01-08:
It seems that commit fails to implement the appropriate **response** header as per https://tools.ietf.org/html/rfc3875#section-6

jow-, 2020-01-09:
The `Content-Type` response header is completely unrelated to your issue and sending a `Content-Type: application/octet-stream` is perfectly valid. Note in your screenshots that the binary garbage you receive starts with `?ELF...` which means that lighttpd failed to execute the requested executable but instead opted to send its file contents instead.

This is a bug in nginx' or uwsgi's configuration or whatever other shenanigans are required to make nginx execute plain old CGI executables in `/cgi-bin/`.

n8v8R, 2020-01-09:
The advise on the *lighttpd* forum been different <sup>**[1]**</sup>

>Your CGI (/cgi-bin/cgi-exec) must respond with appropriate response headers.
See https://tools.ietf.org/html/rfc3875 The Common Gateway Interface (CGI) Version 1.1
___
>This is a bug in lighttpd's configuration. LuCI assumes that executables in /www/cgi-bin/ are executable as CGI programs via http://hostname/cgi-bin/....

I do not think so, the config for LuCI in *lighttpd* reads (and works perfectly fine otherwise)

```
# Set CGI paths
cgi.assign += (
        "/cgi-bin/luci" => ""
)

## Set aliases to LuCI install directory
alias.url += (
        "/cgi-bin/" => "/www/cgi-bin/",
        "/luci-static/" => "/www/luci-static/"
)
```
___
> sending a `Content-Type: application/octet-stream` is perfectly valid

Responding with such header is valid of course but the MIME type `octet-stream` is not valid for `/cgi-bin/cgi-exec`

As stipulated by rfc3875 <sup>**[2]**</sup>

    **The Content-Type response field sets the Internet Media Type [6] of
    the entity body.

      Content-Type = "Content-Type:" media-type NL

    If an entity body is returned, the script MUST supply a Content-Type
    field in the response. If it fails to do so, the server SHOULD NOT
    attempt to determine the correct content type. The value SHOULD be
    sent unmodified to the client, except for any charset parameter
    changes.**
___
<sup>**[1]**</sup> https://redmine.lighttpd.net/boards/2/topics/8893
<sup>**[2]**</sup> https://tools.ietf.org/html/rfc3875#section-6.3.1

jow-, 2020-01-09:
I am not sure what you're getting at. The script sends a `Content-Type` header. And the `mime-type` is `application/octet-stream`, not `octet-stream`.

n8v8R, 2020-01-09:
and the response is sent back as `application/octet-stream`. This is perfectly legal and does not involve the webserver at all.

n8v8R, 2020-01-09:
Let us recap for a moment:

* https://github.com/openwrt/luci/commit/f3724e46a5fc33b3a6fc9f935b88e395c746149f introduced in Master

```
Request.post('/cgi-bin/cgi-exec', postdata, {
			headers: { 'Content-Type': 'application/x-www-form-urlencoded' }
		}
```
* since then the issue started to exhibit in Master
* the commit is not present in 19.07 and thus does not exhibit there
* other CGI scripts do not exhibit the issue in Master or 19.07
* there is no apparent misconfiguration for the CGI exec path in the *lighttpd* config
* the advise from the *lighttpd* forum and subsequent  rfc3875 stipulations are rejected as potential solution
* since *lighttpd* is not vanilla OpenWrt the issue is not being investigated in order to find a solution

jow-, 2020-01-09:
Debugged it myself now. It indeed is a lighttpd specific problem. Lighttpd does not follow symlinks in `/www/cgi-bin/`, even when setting `server.follow-symlink = "enable"`, instead it delivers symlinked CGI programs as files which would even be a security issue in some environments.

You need to add the following explicit CGI mappings to make things work:
```
        "/cgi-bin/luci" => "",
        "/cgi-bin/cgi-backup" => "",
        "/cgi-bin/cgi-download" => "",
        "/cgi-bin/cgi-upload" => "",
        "/cgi-bin/cgi-exec" => ""
```

> since then the issue started to exhibit in Master

Yes, because commit f3724e4 started using a new `cgi-exec` applet which wasn't used previously. Due to the inflexible nature of lighttpd's CGI configuration, you need to whitelist every new CGI applet which is started to get used by LuCI. This is exactly the reason why non-standard webserver setups are not supported officially, they break whenever there are changes in LuCI because they do not meed the expected environment behavior.

n8v8R, 2020-01-09:
Thank you for having taking the time to get to the bottom of this.

jow-, 2020-01-09:
Apparently the following is the only needed thing to provide the expected behavior of "execute anything in /cgi-bin/ as CGI":
```
server.modules += ( "mod_cgi" )
$HTTP["url"] =~ "^/cgi-bin" {
    cgi.assign += ( "" => "" )
}
```

I'd argue that this should even be the default contents of `/etc/lighttpd/conf.d/30-cgi.conf`

n8v8R, 2020-01-09:
For posterity feedback from the *lighttpd* folks

> `server.follow-symlink = "enable"` follows symlinks, as expected.
> The path _/cgi-bin_ is not magical.
> Plain and simple: configuring `cgi.assign` is required for _lighttpd_ to use _mod_cgi_ to handle the request.