openwrt/luci #3432: luci-app-transmission message_level setting not work at transmission-daemon-openssl

Mosney, 2019-12-25:
OpenWrt 19.07rc2, I tried to check/uncheck enable with save&apply and also tried `service transmission restart/stop&start`, not work.

In addition, even other option can be modified normally, but still need manual restart service, it is a feature? I don't know why not reload/restart transmission after luci config change save&apply.
  
  
  
Some detail below:
```
root@OP19:~# ps |grep transmission |grep -v grep
23540 transmis 54580 SN   /usr/bin/transmission-daemon -f --log-error -g /etc/transmission
root@OP19:~# cat /etc/config/transmission |grep message
        option message_level '3'
root@OP19:~# cat /etc/transmission/settings.json |grep message
    "message-level": 1,
```
```
root@OP19:~# opkg list-installed |grep transmission
luci-app-transmission - git-19.356.80404-43c74d3-1
luci-i18n-transmission-zh-cn - git-19.356.80404-43c74d3-1
transmission-daemon-openssl - 2.94-9
transmission-web-control - 2019-04-16-a747a87b-1

```

Mosney, 2019-12-25:
I noticed init.d script use `--log-error`, maybe it is the reason? Should I move this issue to [openwrt/packages](https://github.com/openwrt/packages)?

https://github.com/openwrt/packages/blob/5a25cf86ea56d2019ef551b727918c2e33427502/net/transmission/files/transmission.init#L117

feckert, 2019-12-29:
Please explain what the problem with the option in the init script `--log-error` is. If there is something wrong that is not related to LuCI then we should open an issue in the package feed and close this one.

feckert, 2019-12-29:
Which options can be set?

Mosney, 2019-12-29:
I'm not sure, only from my experience, there is no way to control transmission's log level, and most of the time it will be make log spamming. I mean, 'Message level' setting in luci-app not work, and I reviewed the init script found it will always add '--log-error' parameter. So I just doubt it.

Mosney, 2019-12-29:
So if my guess is correct,it may the packages issue. But vice versa, it may the luci app issue.

Mosney, 2020-01-01:
close due to https://github.com/openwrt/packages/issues/10919#issuecomment-570035453

closed 2020-01-01 by Mosney