openwrt/luci #3433: make Error

snakwu, 2019-12-26:
SHELL= flock /home/snakwu/openwrt-master/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/linux-x86_64/grub-2.04/.pkgdir/grub2-editenv/. /home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.grub2-editenv_installed
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/package/boot/grub2'
time: package/boot/grub2/compile#0.22#0.11#0.41
make[3]: Entering directory '/home/snakwu/openwrt-master/openwrt/package/boot/uboot-envtools'
mkdir -p /home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/snakwu/openwrt-master/openwrt/tmp/.root-copy.flock -c 'cp -fpR /home/snakwu/openwrt-master/openwrt/build_dir/target-x86_64_musl/u-boot-2019.07/.pkgdir/uboot-envtools/. /home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/root-x86/'
touch /home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/root-x86/stamp/.uboot-envtools_installed
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/package/boot/uboot-envtools'
time: package/boot/uboot-envtools/compile#0.23#0.24#0.73
make[3]: Entering directory '/home/snakwu/openwrt-master/openwrt/feeds/luci/modules/luci-base'
touch /home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/luci-base/.configured
touch: cannot touch '/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/luci-base/.configured': No such file or directory
Makefile:42: recipe for target '/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/luci-base/.configured' failed
make[3]: *** [/home/snakwu/openwrt-master/openwrt/build_dir/hostpkg/luci-base/.configured] Error 1
make[3]: Leaving directory '/home/snakwu/openwrt-master/openwrt/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.83#0.92#3.15
package/Makefile:111: recipe for target 'package/feeds/luci/luci-base/host/compile' failed
make[2]: *** [package/feeds/luci/luci-base/host/compile] Error 2
make[2]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
package/Makefile:107: recipe for target '/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/snakwu/openwrt-master/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/snakwu/openwrt-master/openwrt'
/home/snakwu/openwrt-master/openwrt/include/toplevel.mk:225: recipe for target 'world' failed
make: *** [world] Error 2
snakwu@hassio-server:~/openwrt-master/openwrt$

Josua-SR, 2019-12-26:
I also have this issue. My work-around:

```
diff --git a/modules/luci-base/Makefile b/modules/luci-base/Makefile
index ca7de0b..c84d53d 100644
--- a/modules/luci-base/Makefile
+++ b/modules/luci-base/Makefile
@@ -16,6 +16,8 @@ LUCI_DEPENDS:=+lua +luci-lib-nixio +luci-lib-ip +rpcd +libubus-lua +luci-lib-jso
 
 PKG_LICENSE:=MIT
 
+HOST_BUILD_DIR = $(BUILD_DIR_HOST)/$(PKG_NAME)
+
 include $(INCLUDE_DIR)/host-build.mk
 
 define Package/luci-base/conffiles
```

1Jeff1, 2019-12-26:
> I also have this issue. My work-around:
>
> ```
> diff --git a/modules/luci-base/Makefile b/modules/luci-base/Makefile
> index ca7de0b..c84d53d 100644
> --- a/modules/luci-base/Makefile
> +++ [...]

@Josua-SR Thanks that fix works.

snakwu, 2019-12-26:
> I also have this issue. My work-around:
>
> ```
> diff --git a/modules/luci-base/Makefile b/modules/luci-base/Makefile
> index ca7de0b..c84d53d 100644
> --- a/modules/luci-base/Makefile
> +++ [...]

Thanks

klukonin, 2019-12-26:
Worked for me too +1

closed 2019-12-26 by hnyman

NeoRaider, 2019-12-27:
The underlying issue seems to be the following: `include/host-build.mk` completely breaks when HOST_BUILD_DIR is modified after its inclusion. While I didn't look into this in detail yet, I assume that including `luci.mk` after `host-build.mk` is the actual cause, as it will define `PKG_VERSION` and thus implicitly modify the default HOST_BUILD_DIR value.

I'll have another look at this tomorrow.

snakwu, 2019-12-27:
> The underlying issue seems to be the following: `include/host-build.mk` completely breaks when HOST_BUILD_DIR is modified after its inclusion. While I didn't look into this in detail yet, I assume [...]

ok,thanks,I tried to fix the problem, but still reported an error, libubox 2017xx, and couldn't find the library file.

hnyman, 2019-12-27:
> but still reported an error, libubox 2017xx, and couldn't find the library file.

Not related. You have dirty build env. 
Do make clean, update sources and feeds. Then build again.

snakwu, 2019-12-27:
> > but still reported an error, libubox 2017xx, and couldn't find the library file.
> 
> Not related. You have dirty build env.
> Do make clean, update sources and feeds. Then build again.

Okay, it has been compiled successfully, but many network related plugins are wrong!
/usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.model.network' from file '/usr/lib/lua/luci/model/network.lua':
	/usr/lib/lua/luci/model/network.lua:1485: ')' expected near '!'
stack traceback:
	[C]: in function 'loader'
	/usr/lib/lua/luci/ccache.lua:70: in function </usr/lib/lua/luci/ccache.lua:61>
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/nlbw/config.lua:8: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1232: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:916: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:440: in function </usr/lib/lua/luci/dispatcher.lua:439>

hnyman, 2019-12-27:
>  but many network related plugins are wrong!
> /usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.model.network' from file '/usr/lib/lua/luci/model/network.lua':
> /usr/lib/lua/luci/model/network.lua:1485: ')' expected near '!'

Unrelated to the make error.

Caused by #1951 , fix proposed in #3434

hnyman, 2019-12-27:
@NeoRaider 

> The underlying issue seems to be the following: include/host-build.mk completely breaks when HOST_BUILD_DIR is modified after its inclusion. While I didn't look into this in detail yet, I assume [...]

> I'll have another look at this tomorrow

Great. 
I briefly looked at the related .mk files, but did not immediately figure out the correct approach, so I implemented the quick fix yesterday.

This is an architecture question, maybe for @jow- , but I wonder if the remaining host tools (po2lmo, jsmin) really belong to the luci-base package itself, or should they be defined either as normal build tools in OpenWrt, or as a separate luci-buildtools package, maybe with a "normal Makefile" instead of being the host build section of luci-base.

NeoRaider, 2019-12-27:
There is no trivial fix, as:
- host build fails when host-build.mk is included before luci.mk, as luci.mk modifies PKG_VERSION
- host pkginfo dump is missing host build info when host-build.mk is included after luci.mk, as luci.mk calls `BuildPackage` before BUILD_TYPES has been set up

With the way luci.mk currently works there is no clean way to use the same package Makefile for host builds. I like the idea of splitting out the LuCI buildtools into a separate package.

I see a few possible solutions for making luci.mk more compatible with host-build.mk. I think I'd prefer the following, but this will require modifying all Makefiles using luci.mk:

Instead of calling BuildPackage implicitly from luci.mk, this could be moved into a separate define like `BuildPackageLuCI` - that would replace the `# call BuildPackage - OpenWrt buildroot signature` hack. Thoughts? @hnyman @jow-

commit 1ccb8e10 (2019-12-26): luci-base: revert removal of HOST_BUILD_DIR definition

PR #3352 removed the luasrcdiet subpackage definition from
luci-base Makefile, and it also removed the HOST_BUILD_DIR
definition, which looked like it was related to only luasrcdiet,
but apparently that HOST_BUILD_DIR is needed also for the main
luci-base package.

Fixes #3433