openwrt/luci #3435: openwrt-imagebuilder-ar71xx-tiny: Cannot satisfy the following dependencies for luci-app-firewall

strasharo, 2019-12-27:
So I pulled the imagebuilder from here:
https://downloads.openwrt.org/snapshots/targets/ar71xx/tiny/openwrt-imagebuilder-ar71xx-tiny.Linux-x86_64.tar.xz

And tried to build with the following commandline:
`make clean;make image PROFILE=tl-wr703n-v1 PACKAGES="uhttpd uhttpd-mod-ubus libiwinfo-lua luci-base luci-app-firewall luci-mod-admin-full luci-theme-bootstrap kmod-rt2800-lib kmod-rt2800-usb rt2800-usb-firmware kmod-usb-storage kmod-fs-ext4 block-mount -ppp -ppp-mod-pppoe -ip6tables -odhcp6c -kmod-ipv6 -kmod-ip6tables -odhcpd-ipv6only"`

However I got the following exception:
```
Configuring uhttpd.
Configuring uhttpd-mod-ubus.
Configuring kmod-ipt-offload.
Configuring urngd.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-firewall:
 * 	libip4tc2
 * 	libip6tc2
 * opkg_install_cmd: Cannot install package luci-app-firewall.
make[2]: *** [Makefile:156: package_install] Error 255
make[1]: *** [Makefile:117: _call_image] Error 2
make: *** [Makefile:196: image] Error 2

```

So I tried removing luci-app-firewall from the list of packages and ran with this command line:
`make clean;make image PROFILE=tl-wr703n-v1 PACKAGES="uhttpd uhttpd-mod-ubus libiwinfo-lua luci-base luci-mod-admin-full luci-theme-bootstrap kmod-rt2800-lib kmod-rt2800-usb rt2800-usb-firmware kmod-usb-storage kmod-fs-ext4 block-mount -ppp -ppp-mod-pppoe -ip6tables -odhcp6c -kmod-ipv6 -kmod-ip6tables -odhcpd-ipv6only"`

But I got the same exception for another package:
```
Configuring kmod-ipt-offload.
Configuring urngd.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for firewall:
 * 	libip4tc2
 * 	libip6tc2
 * opkg_install_cmd: Cannot install package firewall.
make[2]: *** [Makefile:156: package_install] Error 255
make[1]: *** [Makefile:117: _call_image] Error 2
make: *** [Makefile:196: image] Error 2
```
I tried appending these package to the list of packages I want , but that didn't help:
`make clean;make image PROFILE=tl-wr703n-v1 PACKAGES="uhttpd uhttpd-mod-ubus libiwinfo-lua luci-base luci-mod-admin-full luci-theme-bootstrap kmod-rt2800-lib kmod-rt2800-usb rt2800-usb-firmware kmod-usb-storage kmod-fs-ext4 block-mount libip4tc2 libip6tc2 -ppp -ppp-mod-pppoe -ip6tables -odhcp6c -kmod-ipv6 -kmod-ip6tables -odhcpd-ipv6only"
`
Got the same exception:

```
Configuring kmod-ipt-offload.
Configuring urngd.
Collected errors:
 * opkg_install_cmd: Cannot install package libip4tc2.
 * opkg_install_cmd: Cannot install package libip6tc2.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for firewall:
 * 	libip4tc2
 * 	libip6tc2
 * opkg_install_cmd: Cannot install package firewall.
make[2]: *** [Makefile:156: package_install] Error 255
make[1]: *** [Makefile:117: _call_image] Error 2
make: *** [Makefile:196: image] Error 2

```

Am I doing something wrong there or there's some bug involved? I tried following these instructions: https://openwrt.org/docs/guide-user/additional-software/saving_space

hnyman, 2019-12-27:
< Am I doing something wrong there or there's some bug involved?

Neither. It is about time difference in package version info in different source repos and the packages compiled from those.

iproute2 that includes those libip... packages, was updated recently. Either buildbot has still not built the new versions, or more likely, you have outdated imagebuilder that references to already bypassed libip... versions.

You might clean your package lists, or reload the whole imagebuilder.

(I am not an imagebuilder specialist, so no info about the easiest way...)

hnyman, 2019-12-27:
But the issue has nothing to do with LuCI GUI, so I close this.

closed 2019-12-27 by hnyman

strasharo, 2019-12-27:
Which is the right place to submit these,  the packages repo here in Github or https://bugs.openwrt.org/ ?