openwrt/luci #3437: The system log is always loading, and many network-related plugin settings pages have errors!

snakwu, 2019-12-27:
The system log is always loading, and many network-related plugin settings pages have errors!

/usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.model.network' from file '/usr/lib/lua/luci/model/network.lua':
	/usr/lib/lua/luci/model/network.lua:1485: ')' expected near '!'
stack traceback:
	[C]: in function 'loader'
	/usr/lib/lua/luci/ccache.lua:70: in function </usr/lib/lua/luci/ccache.lua:61>
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/nlbw/config.lua:8: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1232: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:916: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:440: in function </usr/lib/lua/luci/dispatcher.lua:439>

/usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.model.network' from file '/usr/lib/lua/luci/model/network.lua':
	/usr/lib/lua/luci/model/network.lua:1485: ')' expected near '!'
stack traceback:
	[C]: in function 'loader'
	/usr/lib/lua/luci/ccache.lua:70: in function </usr/lib/lua/luci/ccache.lua:61>
	[C]: in function 'require'
	/usr/lib/lua/luci/model/cbi/mwan/globalsconfig.lua:4: in function 'func'
	/usr/lib/lua/luci/cbi.lua:66: in function 'load'
	/usr/lib/lua/luci/dispatcher.lua:1232: in function '_cbi'
	/usr/lib/lua/luci/dispatcher.lua:916: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:440: in function </usr/lib/lua/luci/dispatcher.lua:439>

hnyman, 2019-12-27:
> The system log is always loading

Not related to LuCI.
Caused by ubus and/or libubox update yeasterday by @ynezz 
Discussed in http://lists.infradead.org/pipermail/openwrt-devel/2019-December/020840.html

> many network-related plugin settings pages have errors!
>
> /usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.model.network' from file '/usr/lib/lua/luci/model/network.lua':
> /usr/lib/lua/luci/model/network.lua:1485: ')' expected near '!'

Likely caused by commit 72247ee09f17fe1f7  in #1951 that was merged yesterday.
Fix proposed in #3434

snakwu, 2019-12-27:
This is indeed a plug-in problem, but the system log is still loading, this problem has not been fixed

jow-, 2019-12-27:
The system log loading is caused by hanging logread which in turn is caused by libubox updates in OpenWrt master.

closed 2019-12-27 by jow-