openwrt/luci #3442: satisfy_dependencies_for libubus20191219

peterwillcn, 2019-12-28:
```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-compat:
 * 	libubus20191219
 * opkg_install_cmd: Cannot install package luci-compat.
```

hnyman, 2019-12-28:
Temporary glitch, until buildbot has compiled all phase2 packages.
(due to recent ubus version bump )

hnyman, 2019-12-28:
@jow- 
any idea while phase2 buildbot is so slow nowadays?

Before summer it used to compile incremental package builds rather quickly (30-45 minutes per build round), but in August-September(?) it slowed down significantly (and builds now take 6-9 hours) making the completion of a phase2 build round a multi-day task, instead of the earlier "few hours").  Something changed in the build logic (or check if necessary to build logic.)

This causes this kind of bug reports, as the imagebuilder and phase2 user space package repo are out-of-sync with phase1 images for a quite long time after some version bumps.

closed 2019-12-28 by hnyman