openwrt/luci #3454: luci-app-openvpn is not able to upload conf file

mimmoLinino, 2020-01-02:
I am using 18.06.4 release on Linkit target. I build and install on the target the version of luci-app-openvpn at commit 0f8e36f that contains the ovpn configuration upload feature.
The gui reports all the new fields but when I try to enter the ovpn file containing and press the upload button, nothing happens. The page is refreshed but No error, no file copied on target, no vpn profile created.

feckert, 2020-01-03:
Are there any errors reported in the syslog of the Linkit target?

mimmoLinino, 2020-01-03:
No errors are reported on syslog. 
Let me know if I need to enable something in order to help for the debug session.

dibdot, 2020-01-03:
This feature has never been backported to 18.06.x, please retest with forthcoming stable 19.x or master.

mimmoLinino, 2020-01-03:
tried also on 19.x with the same result. ovpn file is not uploaded on 
the target.

dibdot, 2020-01-03:
Please provide the exact openwrt/luci version you've tested and also the obfuscated ovpn you've tried to upload.

mimmoLinino, 2020-01-03:
Following is the structure of the ovpn file I am passing to the luci-app.

I removed the reference to IP and port and all the ceritificate content.

Can you provide a sample of ovpn file that works?

copying manually my ovpn file in /etc/openvpn/my-vpn.conf the file is 
properly loaded in custom_config session and the vpn connection is up.

# ========================
# OpenVPN Client (<NAME>)
# ========================
client
remote <IP1> <PORT1>
remote <IP2> <PORT2>
server-poll-timeout 5
remote-random
resolv-retry 30
proto udp
dev tun
keepalive 10 60
nobind
float
persist-key
persist-tun
remote-cert-tls server
key-direction 1
comp-lzo no
verb 0

script-security 2

<tls-auth>
#
# 2048 bit OpenVPN static key
#
-----BEGIN OpenVPN Static key V1-----

-----END OpenVPN Static key V1-----
</tls-auth>
<ca>
-----BEGIN CERTIFICATE-----

-----END CERTIFICATE-----
</ca>
<cert>

-----BEGIN CERTIFICATE-----

-----END CERTIFICATE-----
</cert>
<key>
-----BEGIN PRIVATE KEY-----

-----END PRIVATE KEY-----
</key>

dibdot, 2020-01-03:
Sorry, I'm unable to reproduce your issue. I've tested the upload with a GL-MT300Nv2 with 19.07rc2 and current 19.07 snapshot, both versions ware working like a charme, e.g. tested with a mullvad ovpn file:

[image]

[image]

[image]

feckert, 2020-01-08:
I have now checked this again and from my side it works as well.
Do have a fresh vanilla openwrt master/opnewrt-19.07 or did you have anything patched?
Also which browser do you use?

camelcamro, 2020-01-11:
for me also not working, 
current trunk: git-20.004.33955-4c4d6e2) / OpenWrt SNAPSHOT r11890-e98e046f06

and also tried to remove a lot of infos which were fine accepted by the normal openvpn when i start via cmd line ... so the ovpn file is vaild ...

maybe anyone can provide a ovpn file, which really works.
so, that i can test again ?

maybe the problem is different ?
eg: after installation, the /etc/openvpn folder is totally empty ... maybe a file missing ?
eg: /etc/openvpn/vpn.conf

camelcamro, 2020-01-11:
i found now 1 mistake on my config ....
parameter: "log-append" needs to be changed to "log_append"
maybe that helps for you as well ....
so, if there is any parameter which is not correct configured or not correct written the whole VPN load process fails...
maybe that is also the reason on your side ?
if so, pls let us know to CLOSE this issues as "INVALID"

mimmoLinino, 2020-01-13:
I don't have that parmeter on my config.

Let me share my conf (removing the sensible data), if you can share also 
the one that it works I can double check if something is wrong and not 
accepted.

client
remote <IP1> <PORT1>
remote <IP2> <PORT2>
server-poll-timeout 5
remote-random
resolv-retry 30
proto udp
dev tun
keepalive 10 60
nobind
float
persist-key
persist-tun
remote-cert-tls server
key-direction 1
comp-lzo no
verb 0
script-security 2

<tls-auth>

</tls-auth>
<ca>

</ca>
<cert>

</cert>
<key>

</key>

Any log to enable in order to check what is wrong?

QuicksilverMachine, 2020-01-20:
I just had the same issue. It seems like the **/etc/openvpn** directory is sometimes missing. 
You can just ssh to the router as your root user and create it manually. After I did that the configuration got uploaded fine.

feckert, 2020-01-21:
This fix from @jow- should maybe fix your issue in the master branch https://github.com/openwrt/luci/commit/cc01770fa1cf09b729dd931df77b149d1b20d2ef

jow-, 2020-01-21:
@feckert - my fixes will likely not take care of creating `/etc/openvpn`

feckert, 2020-01-21:
I have reread the description of the issue.
According to the description an 18.06 is used there and a newer luci-app-openvpn is installed whose version is not described in detail. This use case is not recommended and not supported.
@mimmoLinino Please check with latest master or last supported stable release (openwrt-19.07). The LuCI has to match!
I will close the issue now.
If you @mimmoLinino still have this issue then feel free to open this again.

mimmoLinino, 2020-01-21:
I Florian,

I have already tested with the released version 19.07 on mt86x8 device 
and the issue is still present.

I have to verify with this new commit if it resolve the issue.

Thanks

Mimmo

camelcamro, 2020-01-21:
> I don't have that parmeter on my config. Let me share my conf (removing the sensible data), if you can share also the one that it works I can double check if something is wrong and not accepted. [...]

well, already told that there are few limitation on loading a WRONG config file.. it is not handling all the errros....

pls recognize by yourself what are the wrong entries to find the iussue:
simple comment all lines with a "#", and then try it out ...
and add then more and more, and you'll find the wrong entries...

brnnnrsmssn, 2020-04-07:
LuCI openwrt-19.07 branch (git-20.096.65488-81ad3bb) / OpenWrt 19.07.2 r10947-65030d81f3
linksys ea6350

i believe i am experiencing this same issue.
when i attempt to upload the file (content below) named "config.ovpn" the ui refreshes the page and no instance is added.
all packages that were installed by default have been updated to the latest version available through the luci ui. the luci-app-openvpn package is installed, but no other openvpn packages (easy-rsa or ssl) have been installed. i am attempting to do everything through the ui and avoid needing to connect to the router in other ways.
in the file do all hyphens ("-") need to be replaced by underscores ("_"), or just some of them? shouldn't the process of uploading a configuration file automatically handle correctly parsing the file instead of off-loading that task to the user?

contents of file being uploaded (some unique content removed)
```
client
dev tun
proto udp

remote ...
remote ...
remote ...
remote ...
remote ...

remote-random
resolv-retry infinite
nobind
cipher AES-256-CBC
auth SHA512
comp-lzo no
verb 3

tun-mtu 1500
tun-mtu-extra 32
mssfix 1450
persist-key
persist-tun

reneg-sec 0

remote-cert-tls server
auth-user-pass
pull
fast-io

script-security 2
up /etc/openvpn/update-resolv-conf
down /etc/openvpn/update-resolv-conf

<ca>
-----BEGIN CERTIFICATE-----
...
-----END CERTIFICATE-----
</ca>

key-direction 1
<tls-auth>
-----BEGIN OpenVPN Static key V1-----
...
-----END OpenVPN Static key V1-----
</tls-auth>
```

brnnnrsmssn, 2020-04-07:
even with all hyphens replaced by underscores the result is the same.

dibdot, 2020-04-08:
@feckert Why do we re-open this one? The fix is already in 19.07-branch (https://github.com/openwrt/luci/commit/ed1fc639acbab36ba2ae7d1e12680c67cfa3dfff) and will be included in 19.07.3

feckert, 2020-05-29:
@dibdot sorry my fault

closed 2020-05-29 by feckert

VIIIHenryVIII, 2020-07-20:
> 
> 
> I just had the same issue. It seems like the **/etc/openvpn** directory is sometimes missing.
> You can just ssh to the router as your root user and create it manually. After I did that the configuration got uploaded fine.

I know this is a closed thread but this helped in my case

kpma1985, 2021-04-30:
Same here,

changed from openwrt theme to bootstrap was my resolution on Powered by LuCI openwrt-21.02 branch (git-21.106.55967-06dd6b5) / OpenWrt 21.02.0-rc1 r16046-59980f7aaf

antrecu, 2021-05-25:
> > I just had the same issue. It seems like the **/etc/openvpn** directory is sometimes missing.
> > You can just ssh to the router as your root user and create it manually. After I did that the configuration got uploaded fine.
> 
> I know this is a closed thread but this helped in my case

same here!

antrecu, 2021-05-25:
> > I just had the same issue. It seems like the **/etc/openvpn** directory is sometimes missing.
> > You can just ssh to the router as your root user and create it manually. After I did that the configuration got uploaded fine.
> 
> I know this is a closed thread but this helped in my case

Same Here!!

robi-reavisys, 2021-07-02:
I have same problem on LuCI openwrt-19.07 branch (git-21.044.30835-34e0d65) / OpenWrt 19.07.7 r11306-c4a6851c72 installed on Mikrotik router.
Solution was mkdir /etc/openvpn
If I will be able to reacreate problem, I will update this thread.

vladipus, 2021-07-30:
For me the solution was to install the openvpn-openssl module first.
Doesn't it have to be in the dependencies?

souhaiebtar, 2022-01-01:
> For me the solution was to install the openvpn-openssl module first. Doesn't it have to be in the dependencies?

your comment did help, by installing `openvpn-openssl` upload did work

danilvalov, 2022-12-08:
I had the same problem with uploading ovpn-file.
I found my problem: I used Chrome browser, and my OpenWRT admin panel had untrusted certificate (I just pressed `Open anyway` when Chrome asked me). And Chrome browser just skipped uploading on it side.

I opened the OpenWRT admin panel in Firefox, and my OVPN file was uploaded successful