openwrt/luci #3460: firewall: when editing a zone, "Covered networks" isn't saved

taggart, 2020-01-04:
Running OpenWrt 19.07.0-rc2 r10775-db8345d8e4 / LuCI openwrt-19.07 branch git-19.334.63023-039ef1f

When editing a zone and changing "Covered networks" and then saving and applying, the result isn't saved and the page still lists "unspecified". If I manually add a "option network 'foo'" line to that zone in /etc/config/firewall then I am able to change it in the web interface and it gets updated properly. So I suspect it's due to not having a line initially.

jow-, 2020-01-04:
Were the zone and the network name equal? E.g. zone `foo` with network `foo` ?

taggart, 2020-01-05:
Yes. Let me know if you want me to test different combinations.

taggart, 2020-01-05:
Looks like this was fixed in d60a1a13b091c622bc2d4d63321d2e3ea0437796

jow-, 2020-01-05:
Thanks, I suspected that. Will close this.

closed 2020-01-05 by jow-