openwrt/luci #3464: luci-app-smbd: problem saving data

Ansuel, 2020-01-05:
There is a problem in the app for smbd. I can't save anything as it does respond me with error data not valid. 
I fixed this by removing the template section from the JS so i think there is a problem with the validation of that section.

hnyman, 2020-01-05:
cc @Andy2244

Andy2244, 2020-01-05:
@Ansuel Cant reproduce this, is this in master or 19.07? I just added a new share and changed some options, while also editing the template via luci, all worked fine for me. I also compared all my live files to the github versions and all are identically, so i need more info on this.

Ansuel, 2020-01-05:
master. 
If you don't change anything. Does luci report error or save correctly?

Andy2244, 2020-01-05:
> If you don't change anything. Does luci report error or save correctly?

Not sure what you mean, but i get "No changes to apply" if i don't change anything and if i edit the template i get the same message, which is a bit odd, but the template change is actually written correctly for my.

Ansuel, 2020-01-05:
On my side if I save i have this

Cannot save due to invalid values

By removing template section i don't get this.

Ansuel, 2020-01-05:
I notice now that even with the error... the template file is updated... so i think there is something strange here...

Andy2244, 2020-01-05:
Can you compare the github default template `/etc/smbd/smb.conf.template` with your live version?

Andy2244, 2020-01-05:
Yeah can't remember how the old lua version handled template changes, but i use the same code in samba4, so both apps are affected. 
Need to search for other .js template examples or someone else with more luci .js knowledge need take a look.

Andy2244, 2020-01-05:
mhh can't find any other .js example that uses a `form.TextValue` and the code "looks" fine.
@hnyman maybe a luci .js "expert" can take a quick look?

Ansuel, 2020-01-05:
ok I understand the problem...

.write wants true as a return value  but... 
fs.write doesn't return true

This fixes the problem... don't know if this is intended...
@jow- 

```
o.write = function (section_id, formvalue) {
         fs.write('/etc/smbd/smb.conf.template', formvalue.trim().replace(/\r\n/g, '\n') + '\n');
         return true;
      };
```

Andy2244, 2020-01-05:
I see no difference on my end, but if i change the template while it gets changed, the change is not detected? So the service is not restarted on template changes via save&apply?

jow-, 2020-01-05:
That sounds odd. The `form.js` logic does not care about the return value of `write()`. Maybe the change in behavior was just coincidental and what you actually did was invalidating some browser caches.

Ansuel, 2020-01-05:
I have the same behavior in InPrivate browser.

jow-, 2020-01-05:
The "no changes" message is also expected in the template case since that action modifies no uci.

jow-, 2020-01-05:
My guess is that an exception occurs somewhere in the codeflow, maybe some nginx integration issue again ;) Try open the browsers code debugger and check both "Halt on exceptions" and "Halt on handled exceptions".

Ansuel, 2020-01-05:
Ok tested more... The error is triggered by actually return fs.write ...

Andy2244, 2020-01-05:
> The "no changes" message is also expected in the template case since that action modifies no uci.

ok make sense, so how to handle this case? I think in the past i just manually restarted the service on a template change, but that's rather buggy. So i should trigger a service restart inside the write() logic?

jow-, 2020-01-05:
> ok make sense, so how to handle this case?

Ignore it for now.

Ansuel, 2020-01-05:
@Andy2244 sorry for the mess. Caused by nginx module. Fixed...

closed 2020-01-05 by Ansuel