openwrt/luci #3466: luci-app-unbound: config zone, server string parsing issue

schm0, 2020-01-05:
Hi!
Using the following config:
```
config zone
	list zone_name '.'
	option zone_type 'forward_zone'
	# Quad9 Secure
	list server '9.9.9.9@853#dns.quad9.net'
	list server '149.112.112.112@853#dns.quad9.net'
	list server '2620:fe::fe@853#dns.quad9.net'
	list server '2620:fe::fe:9@853#dns.quad9.net'
	option tls_upstream '1'
	option fallback '0'
	option enabled '1'
```
The final generated string of `list server '2620:fe::fe@853#dns.quad9.net'` in /var/lib/unbound.conf always contains `@853`
For example,
`forward-addr: 2620:fe::fe@853#dns.quad9.net@853`
Because of this the SSL hand shake fails.

Can you add a possibility to the luci unbound setup page to let the user configure the allowed networks (access-control) please?

jow-, 2020-01-07:
@EricLuehrsen

EricLuehrsen, 2020-01-11:
This will be an issue for /openwrt/packages rather than /openwrt/luci. The shell script should detect you already have a port or TLS domain name and not suffix the default.

EricLuehrsen, 2020-01-11:
This will be an issue for /openwrt/packages rather than /openwrt/luci. The shell script should detect you already have a port or TLS domain name and not suffix the default. However in your case, you can just set the addresses alone and the option `tls_index dns.quad9.net` because the standard port is 853 and all your servers are quad9. The UCI scripts will assemble the Unbound options for you.
```
config zone
	list zone_name '.'
	option zone_type 'forward_zone'
	# Quad9 Secure
	option tls_index 'dns.quad9.net'
	list server '9.9.9.9'
	list server '149.112.112.112'
	list server '2620:fe::fe'
	list server '2620:fe::fe:9'
	option tls_upstream '1'
	option fallback '0'
	option enabled '1'
```

EricLuehrsen, 2020-01-11:
As for your access control request, I have kept the Unbound UCI similar in concept to the dnsmasq UCI where it makes sense. For access control like dnsmasq, you can accept any network such as for a large corporate network exit server or you can accept only the subnet(s) directly connected to the router. In both cases, the firewall is what is responsible to prevent external global requests.

Longer term, I plan on using the Unbound view/tag feature within access control to create unique network DNS capability. For example, you could configure the often WiFi "guest" multiplexed SSID to instead be "child protected" SSID and provide a view for it with an extensive block list. The challenge is Unbound's features are rich and complicated, so it takes a lot of time to create a right-sized UCI.

closed 2020-01-11 by hnyman