openwrt/luci #3467: broken wireless info

Ansuel, 2020-01-05:
[image]

hnyman, 2020-01-05:
Where? With what?
Seems to work ok with r11922-2d80f7e836 using normal uhttpd and bootstrap theme.

[image]

Ansuel, 2020-01-05:
Mvebu

hnyman, 2020-01-05:
Works for me with my mvebu WRT3200ACM at r11922-2d80f7e836 with uhttpd and bootstrap theme

[image]

jow-, 2020-01-06:
Works fine for me too on a Mir3G. Your screenshot looks like a memory/ABI corruption though, make sure that `libiwinfo` and `libiwinfo-lua` as well as `rpcd-mod-iwinfo` are in sync and all updated / rebuilt to their latest version.

Ansuel, 2020-01-06:
Aside the fact that is caused by my buildroot... I think there is a problem in detecting abi change by the dependent package. Shouldn't they had to rebuild automatically after a version upgrade?

closed 2020-01-06 by Ansuel