openwrt/luci #3469: "cgi-bin/luci/admin" not the default start anymore?

Andy2244, 2020-01-06:
I noticed that for like 1-2 months now i have to scroll down and click the "Administration" link, after i entered my password, to get access to actual settings tabs?
I assumed this is a bug, but i build from master today and same "issue"? So is the "status" page now the default page after a login and i have to scroll to the Administration link always or is this some bug?

jow-, 2020-01-06:
Uhm, I am not sure what you mean. The status overview page has always been the default

Andy2244, 2020-01-06:
What i mean is i only see the status page, there are no extra tabs visible, only after i click on the "Administration" link are they visible?

hnyman, 2020-01-06:
http://192.168.1.1/cgi-bin/luci/ works quite normally for me with the normal unmodified LuCI, default uhttpd server and default bootstrap theme.
(LuCI Master (git-20.006.34741-55dcc8a) / OpenWrt SNAPSHOT r11924-99dd2709b8 )

Are you using something special?

jow-, 2020-01-06:
Also check the browser debug console for errors and (catched) exceptions. Maybe something is breaking the initial menu rendering.

Andy2244, 2020-01-06:
mhh ok i get this:
```
HTTP404: NOT FOUND - The server has not found anything matching the requested URI (Uniform Resource Identifier).
(XHR)GET - http://192.168.1.1/luci-static/resources/protocol/relay.js?v=git-19.292.31773-cc35206
```
Yes there is no relay.js there? I used the same base diffconfig i have for years, so whats changed in the last 2 months and how do i fix this?

Andy2244, 2020-01-06:
mhh ok i installed the relayd + luci-proto-relay and now the error is gone, but i still land on a page without the settings?
I also see no errors/warnings now at all?

jow-, 2020-01-06:
The 404 is expected and harmless. Try opening tge JS debugger tab and check “halt on exceptions” and “halt on catched exceptions” then see which one occur. There are one or two expected ones like a class not found error, but maybe there’s other ones like a null access.

Andy2244, 2020-01-06:
> Try opening tge JS debugger tab and check “halt on exceptions” and “halt on catched exceptions” then see which one occur.

I don't get any exceptions this way?

jow-, 2020-01-06:
Then I’m out of ideas unfortunately. I am unable to reproduce the described issue either 😕

Andy2244, 2020-01-06:
ok if its only me than i have to live with it, i can get the settings via the admin link, was just strange that i have to-do this now always.

closed 2020-01-06 by Andy2244

hnyman, 2020-01-06:
Now that you have added the screenshot to the old message, one can see that the overview contents look ok, but the menus are almost non-existent. 

[image]

There is no "Flash Firmware" menu item in the normal LuCI (as it would be "Backup / Flash Firmware"), I guess that you have some code modifications somewhere?
Or are you really running just the standard vanilla LuCI built from a clean feed repo?

Hmmm....  Based your screenshot I got one idea:

The mini version LuCI "luci-mod-admin-mini" is the only place where the standard LuCI has "Flash Firmware" as a menu item.
https://github.com/openwrt/luci/blob/c32097d7d525d0c3e960e558d1c561c9c596bf96/modules/luci-mod-admin-mini/luasrc/controller/mini/system.lua

Please check that you have not mixed that mini LuCI to your build.

You might show the diffconfig output of your luci items in your .config

Andy2244, 2020-01-06:
> lease check that you have not mixed that mini LuCI to your build.

Have no luci-mod-admin-mini installed, here is my luci config and no i have no modded luci stuff installed.

```
CONFIG_PACKAGE_luci=y
CONFIG_PACKAGE_luci-app-adblock=y
CONFIG_PACKAGE_luci-app-commands=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_luci-app-diag-core=y
CONFIG_PACKAGE_luci-app-firewall=y
CONFIG_PACKAGE_luci-app-ntpc=y
CONFIG_PACKAGE_luci-app-opkg=y
CONFIG_PACKAGE_luci-app-samba4=y
CONFIG_PACKAGE_luci-app-smbd=y
CONFIG_PACKAGE_luci-app-sqm=y
CONFIG_PACKAGE_luci-app-wol=y
CONFIG_PACKAGE_luci-base=y
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-lib-ip=y
CONFIG_PACKAGE_luci-lib-ipkg=y
CONFIG_PACKAGE_luci-lib-jsonc=y
CONFIG_PACKAGE_luci-lib-nixio=y
CONFIG_PACKAGE_luci-mod-admin-full=y
CONFIG_PACKAGE_luci-mod-failsafe=y
CONFIG_PACKAGE_luci-mod-network=y
CONFIG_PACKAGE_luci-mod-status=y
CONFIG_PACKAGE_luci-mod-system=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-proto-ppp=y
CONFIG_PACKAGE_luci-theme-bootstrap=y 
```

hnyman, 2020-01-06:
It is luci-mod-failsafe:

https://github.com/openwrt/luci/blob/master/modules/luci-mod-failsafe/luasrc/controller/failsafe/failsafe.lua

There is the same the menu items that you see.

Andy2244, 2020-01-06:
> There is the same the menu items that you see.

yeah i always had this installed, so why it is now the default page after i login?

jow-, 2020-01-06:
Because the menu building logic was refactored and it works slightly differently now, so the failsafe menu structure happened to become the new default.

I think the behavior was always somewhat undefined as there’s been reports in the past stating that installing mod-failsafe will break the normal menu structure.

As far as I remember, the failsafe mod was never meant to be included in normal builds but to provide a ui in special failsafe mode inages. @dangowrt might recall it in more detail as he introduced it initially.

Andy2244, 2020-01-06:
ok makes sense, so i will just remove it in my next build, thanks.