openwrt/luci #3470: RPC IP call not working

jacekpaszkowski, 2020-01-06:
version: OpenWrt SNAPSHOT r11890-e98e046f06

In latest version i'm not able to call JSONRPC methods, it worked in snapshot version from september 2019.

eg.
http://192.168.x.x/cgi-bin/luci/rpc/ip
```
{"method": "neighbors", "params": [{"family": 4}]}
```

response:
```
{
"id": null,
"error": {
"message": "Parse error.",
"code": -32700
},
"jsonrpc": "2.0"
}
``

Only auth method works correctly.

closed 2020-01-07 by jow-

jacekpaszkowski, 2020-01-07:
@jow- nice to see that the issue is fixed. When there will be new build available of snapshot version ramips/mt7621/?
Sould i update luci packages or do sysupgrade?

jow-, 2020-01-07:
Not sure when new builds will be available but when they are, an `opkg update; opkg upgrade luci-base; rm -r /tmp/luci-*` should be enough to get things going again.

commit 38c9c9e0 (2020-01-07): luci-base: dispatcher: fix rpc controller regression

When testing the luci-rpc authnetication, avoid clobbering the HTTP
post request body.

Fixes: #3470
commit 0f83d819 (2020-05-07): luci-base: dispatcher: fix rpc controller regression

When testing the luci-rpc authnetication, avoid clobbering the HTTP
post request body.

Fixes: #3470

(cherry picked from commit 38c9c9e0a2564b9644a56f1143f6acf0c9e272c8)