openwrt/luci #3471: UPnP Not Working 19.07+

willowen100, 2020-01-07:
I can confirm that UPnP doesn't work. I tried the official stock openwrt firmware version 19.07.0-rc2 and UPnP didn't show any entries. I have also tried compiling from master and that does the same thing. As soon as I flashed 18.06.5 LuCI-App-UPnP worked straight away.

My Router: WRT1900ACSv2
Test devices/software: Sony PS3, Sony PS4, Windows qBitorrent

solidus1983, 2020-01-07:
Its also not working in Master either. have noticed this with OpenWrt SNAPSHOT r11888-32e4eaef1b

jow-, 2020-01-09:
I doubt that it is a LuCI issue though. In my local tests I was unable to reproduce the problem in the first place. Entries showed up when using `upnpc` on a Linux box to manually request forwards, and starting `transmission-gtk` also immediately triggered port forwards for Bittorrent.

Maybe the package feed changed the `miniunpd` daemon in way that it became incompatible with certain devices, but that issue should be raised here then: https://github.com/openwrt/packages/issues?utf8=%E2%9C%93&q=miniupnpd+is%3Aissue

Maybe it is also a problem with multicast traffic and mwlwifi...

solidus1983, 2020-01-09:
Managed to get it working again by deleting the content of upnpd in /etc/config folder and just resetting it all back up. I was mearly point to the person above I had the same issue on the master branch.

But thanks for your input jow.

manjuprajna, 2020-01-10:
I also have same problem, have to check enable UPnP NAT-PMP service, than it showes active upnp redir again. mine is a WRT32x but I think nothing to do with mwlwifi since I am using utorrent on desktop with ethernet cable.

jow-, 2020-01-10:
Seems to be a configuration issue.

closed 2020-01-10 by jow-

willowen100, 2020-01-10:
Sorry for the late reply back. I have empty the entire contents of /etc/config and then restarted the miniupnpnd service. List still empty.

I have noticed that the settings in more recents builds are simply removed when the setting is the default value, rather than keeping the setting in the config and just changing the value

E.g. 
`option enabled '1'
	option enable_upnp '1'
	option enable_natpmp '1'
	option igdv1 '1'`

With this inside the upnpd file if I go back to LuCI and apply changes without actually making any in changes in LuCI straight away UCI removes the parameters.

willowen100, 2020-01-10:
[image]

jow-, 2020-01-11:
LuCI removes these options because they're the default value:
https://github.com/openwrt/packages/blob/master/net/miniupnpd/files/miniupnpd.init#L135

willowen100, 2020-01-11:
I've literally just flashed the new 19.07.0 update compiled by myself with the LuCI-App-UPnP package and tried clearing the /etc/config/upnpd and nothing works. What else can I try?

moralrebuild, 2020-02-28:
still not work on 19.07.1. Please fix this minor bug ASAP

willowen100, 2020-03-02:
I can say that I managed to flash the v18.06.1 version of UpNP into v19.07.1 firmware and it works perfectly. There is definitely something up with the new builds of Miniupnpd.