openwrt/luci #3487: Dependency error - libiwinfo20200105

matadams91, 2020-01-09:
I'm getting error while installing Luci from package http://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/luci_git-20.008.20448-a8087bd-1_all.ipk.
Before installation I made repo update with opkg update, but it still won't work.

Collected errors: 
* satisfy_dependencies_for: Cannot satisfy the following dependencies for Lucy: 
*      libiwinfo20200105

I also tried to use another package from the other version: http://downloads.openwrt.org/snapshots/packages/mipsel_24kc_24kf/
Same results.
Does anyone have similar issues and found a solution for it?

feckert, 2020-01-09:
The package libiwinfo is managed in the package feed.
To make the packages fit together you should install a new image for your hardware and then install the Luci from there. Then it should fit together again. In a stable release this should work the way you did it. But the master is very fragile and something changes all the time.

jow-, 2020-01-09:
Not a LuCI issue.

closed 2020-01-09 by jow-