openwrt/luci #3499: Wireguard interface dialogue

y-alexey, 2020-01-12:
Missing network mask configuration option.

That's not matter, if you using only one upstream connection. But in case of multiple income tunnels, or when try to configure interface for bgp, I need to add mask option for this interface manually by _/etc/config/network_ file editing.

jow-, 2020-01-12:
Which option is missing where exactly?

y-alexey, 2020-01-12:
Interface mask option
`option netmask '255.255.255.0'` for example.

jow-, 2020-01-12:
The underlying OpenWrt protocol handler does not specify a netmask option, instead you‘re supposed to use CIDR or address/mask notation, e.g. `192.168.1.1/24` or `192.168.1.1/255.255.255.0`

closed 2020-01-12 by jow-

y-alexey, 2020-01-12:
Thank you for your time. But I not really understand how to report issue to openwrt.
Wireguard case it's only one.