openwrt/luci #3500: luci-app-nlbwmon: group by source IP instead of MAC ?

erdoukki, 2020-01-13:
How can I have the display sorted by source IP instead of by MAC ?
With a bridge interface, all MAC are unkown, and then the traffic is shown in a pop-up instead of the default page view...
If the group was on Source IP, this behaviour may be bypassed !

Can we have an option to get the group shown by source IP instead of MAC ?
Or can we have the unkown MAC address being shown in the default page ?

Thanks in advance

jow-, 2020-01-13:
This is not supported by nlbwmon.

closed 2020-01-13 by jow-

erdoukki, 2020-01-13:
the console view shows correctly, not the luci view :
`root@v-monitor:~# nlbw -c show
  Fam            Host (    MAC )      Layer7      Conn.   > Downld. ( > Pkts. )      Upload (   Pkts. )
IPv4       10.3.2.128 (00:00:00)       HTTPS       115    309.67 MB ( 283.31 K)    23.71 MB ( 203.14 K)
IPv4       10.3.2.238 (51:82:18)       HTTPS       203    300.35 MB ( 443.36 K)    50.66 MB ( 394.55 K)
IPv4       10.3.2.231 (00:00:00)       HTTPS        33     97.79 MB ( 105.46 K)    64.39 MB ( 101.00 K)
IPv4       10.3.2.236 (00:00:00)        HTTP        18     64.75 MB (  44.90 K)     1.71 MB (  33.55 K)
IPv4       10.3.2.238 (51:82:18)       IMAPS         7     34.19 MB ( 105.18 K)    10.64 MB ( 143.41 K)
IPv4       10.3.2.229 (00:00:00)       HTTPS         6     31.12 MB (  28.14 K)     4.30 MB (  24.69 K)
IPv4       10.3.2.231 (00:00:00)       other        55     16.61 MB (  24.02 K)     6.78 MB (  43.71 K)
IPv4       10.3.2.128 (00:00:00)       IMAPS         4      9.94 MB (  52.21 K)     4.55 MB (  58.59 K)
IPv4       10.3.2.126 (00:00:00)       HTTPS         8      3.08 MB (   5.64 K)     1.63 MB (   5.46 K)
IPv4        10.3.2.51 (00:00:00)        HTTP        21      1.30 MB (     989 )    43.44 KB (     793 )
IPv4       10.3.2.238 (51:82:18)        HTTP        17    809.00 KB (     679 )    39.18 KB (     630 )
IPv4       10.3.2.126 (00:00:00)       other         2    654.06 KB (   3.39 K)     1.58 MB (   3.24 K)
IPv4       10.3.2.229 (00:00:00)       other        16    537.48 KB (   3.02 K)   782.02 KB (   5.15 K)
`

erdoukki, 2020-01-13:
https://raw.githubusercontent.com/erdoukki/vrac/master/2020-01-13_14-07.png

erdoukki, 2020-01-15:
I am looking for this ;
```
root@monitor:~# nlbw -c show -g ip
             IP      Conn.   > Downld. ( > Pkts. )      Upload (   Pkts. )
     10.2.1.242     1.39 K   296.16 MB ( 209.19 K)     9.90 MB ( 182.82 K)
     10.2.1.103     1.36 K    31.77 MB (  36.63 K)     2.88 MB (  26.15 K)
     10.2.1.165       253     26.34 MB (  18.66 K)   575.69 KB (  10.45 K)
     10.2.1.115       169     22.40 MB ( 118.27 K)    24.05 MB ( 128.71 K)
     10.2.1.116       475     14.06 MB (  16.75 K)     7.79 MB (  18.93 K)
       10.2.1.2       730      7.57 MB (  12.11 K)     1.69 MB (  10.44 K)
     10.2.1.241     3.65 K     3.31 MB (   5.63 K)     1.01 MB (  16.79 K)
     10.2.1.164        72      3.01 MB (   2.61 K)   213.27 KB (   2.34 K)
     10.2.1.207       597      2.50 MB (   6.90 K)     1.59 MB (   8.75 K)
      10.2.1.55         8      6.97 KB (      14 )   386.97 KB (  11.98 K)
     10.2.1.125       605      2.61 KB (      10 )    73.40 KB (   1.19 K)
     10.2.1.179       608      2.61 KB (      10 )    73.81 KB (   1.20 K)
     10.2.1.237       607      2.61 KB (      10 )    74.05 KB (   1.20 K)
       10.2.1.3        19      1.41 KB (      19 )     1.41 KB (      19 )
     10.2.1.111         5          0 B (       0 )    26.32 KB (     358 )
     10.2.1.150         9          0 B (       0 )   108.61 KB (   2.76 K)
    10.90.90.90         1          0 B (       0 )    26.59 KB (     296 )
```
instead of this ;
```
root@monitor:~# nlbw -c show -g mac
              MAC      Conn.   > Downld. ( > Pkts. )      Upload (   Pkts. )
00:00:00:00:00:00    10.22 K   380.83 MB ( 408.19 K)    49.90 MB ( 413.97 K)
f0:ad:4e:08:ab:43       253     26.34 MB (  18.66 K)   575.69 KB (  10.45 K)
```

erdoukki, 2020-01-15:
> This is not supported by nlbwmon.

Do you want to say that the beahviour is not nlbwmon ?
I am looking in the luci-app-nlbwmon, and I am sure it is possible...
May be it is in there that I have to test to modify ?

Or may be the mac address detect of nlbwmon is in fault, and may be enhanced to get mac address of "other" 00:00:00:00:00:00 being detected correctly ?

erdoukki, 2020-01-15:
root@monitor:~# diff /www/luci-static/resources/view/nlbw.js.orig /www/luci-static/resources/view/nlbw.js -u
```
--- /www/luci-static/resources/view/nlbw.js.orig	2020-01-15 16:50:49.750440744 +0100
+++ /www/luci-static/resources/view/nlbw.js	2020-01-15 17:26:42.333149287 +0100
@@ -116,7 +116,7 @@
 	var keys = [], columns = {}, records = {}, result = [];
 
 	if (typeof(group) !== 'function' && typeof(group) !== 'object')
-		group = ['mac'];
+		group = ['ip'];
 
 	for (var i = 0; i < trafficData.columns.length; i++)
 		columns[trafficData.columns[i]] = i;
@@ -199,7 +199,7 @@
 
 function fetchData(period)
 {
-	XHR.get(L.url('admin/nlbw/data'), { period: period, group_by: 'family,mac,ip,layer7', order_by: '-rx_bytes,-tx_bytes' }, function(xhr, res) {
+	XHR.get(L.url('admin/nlbw/data'), { period: period, group_by: 'family,ip,mac,layer7', order_by: '-rx_bytes,-tx_bytes' }, function(xhr, res) {
 		if (res !== null && typeof(res) === 'object' && typeof(res.columns) === 'object' && typeof(res.data) === 'object')
 			trafficData = res;
 
@@ -266,7 +266,7 @@
 
 	var detailData = query(
 		function(c, r) {
-			return ((r[c.mac] === key || r[c.ip] === key) &&
+			return ((r[c.ip] === key || r[c.ip] === key) &&
 			        (r[c.rx_bytes] > 0 || r[c.tx_bytes] > 0));
 		},
 		[col],
@@ -337,22 +337,16 @@
 	pie(tooltip.querySelector('#bubble-pie1'), rxData);
 	pie(tooltip.querySelector('#bubble-pie2'), txData);
 
-	var mac = key.toUpperCase();
-	var name = hostInfo.hasOwnProperty(mac) ? hostInfo[mac].name : null;
+	var ip = key.toUpperCase();
+	var name = hostInfo.hasOwnProperty(ip) ? hostInfo[ip].name : null;
 
 	if (!name)
 		for (var i = 0; i < detailData.length; i++)
 			if ((name = hostNames[detailData[i].ip]) !== undefined)
 				break;
 
-	if (mac !== '00:00:00:00:00:00') {
-		kpi(tooltip.querySelector('#bubble-hostname'), name);
-		kpi(tooltip.querySelector('#bubble-vendor'), oui(mac));
-	}
-	else {
-		kpi(tooltip.querySelector('#bubble-hostname'));
-		kpi(tooltip.querySelector('#bubble-vendor'));
-	}
+	kpi(tooltip.querySelector('#bubble-hostname'));
+	kpi(tooltip.querySelector('#bubble-vendor'));
 
 	var rect = this.getBoundingClientRect(), x, y;
 
@@ -417,7 +411,7 @@
 		function(c, r) {
 			return (r[c.rx_bytes] > 0 || r[c.tx_bytes] > 0);
 		},
-		['mac'],
+		['ip'],
 		//function(c, r) {
 		//	return (r[c.mac] !== '00:00:00:00:00:00') ? r[c.mac] : r[c.ip];
 		//},
@@ -431,7 +425,7 @@
 	for (var i = 0; i < hostData.length; i++) {
 		var rec = hostData[i],
 		    mac = rec.mac.toUpperCase(),
-		    key = (mac !== '00:00:00:00:00:00') ? mac : rec.ip,
+		    key = rec.ip,
 		    dns = hostInfo[mac] ? hostInfo[mac].name : null;
 
 		var cell = E('div', formatHostname(dns));
@@ -439,15 +433,18 @@
 		rows.push([
 			cell,
 			E('a', {
-				'href':         '#' + rec.mac,
+				'href':         '#' + rec.ip,
 				'data-col':     'ip',
 				'data-tooltip': _('Source IP')
-			}, (mac !== '00:00:00:00:00:00') ? mac : _('other')),
+			}, 
+			rec.ip
+			),
 			E('a', {
-				'href':         '#' + rec.mac,
+				'href':         '#' + rec.ip,
 				'data-col':     'layer7',
 				'data-tooltip': _('Protocol')
-			}, '%1000.2m'.format(rec.conns)),
+			}, 
+			'%1000.2m'.format(rec.conns)),
 			'%1024.2mB'.format(rec.rx_bytes),
 			'%1000.2mP'.format(rec.rx_pkts),
 			'%1024.2mB'.format(rec.tx_bytes),
@@ -608,10 +605,7 @@
 
 	var rows = [];
 
-	for (var mac in records) {
-		if (mac === '00:00:00:00:00:00')
-			continue;
-
+	for (var ip in records) {
 		var dns = hostInfo[mac] ? hostInfo[mac].name : null,
 		    rec4 = records[mac][4],
 		    rec6 = records[mac][6];
```

erdoukki, 2020-01-15:
[image]