openwrt/luci #3501: The word "Load" has two different meanings but the same translation

urbalazs, 2020-01-13:
The word "Load" has two different meanings:

- Load as a button text to load something (as a verb)
- Load as heading or title to display system load (as a noun)

Weblate has a builtin [translation consistency check](https://docs.weblate.org/en/weblate-3.10/admin/continuous.html#translation-consistency), that makes it possible to translate a string only one time, and Weblate will copy the translation into all components. In this case it is not possible to add different translations for *load* as a verb and *load* as a noun.

Possible solution would be to improve the wording, like:

- Load as verb to *Load something* (e. g. *Load data*)
- Load as noun to *System Load*

Affected strings:

- Load -> Load data
  https://github.com/openwrt/luci/blob/master/applications/luci-app-banip/luasrc/view/banip/ipsetview.htm#L59
  https://github.com/openwrt/luci/blob/master/applications/luci-app-banip/luasrc/view/banip/ripeview.htm#L97

- Load -> System Load
  https://github.com/openwrt/luci/blob/master/modules/luci-mod-status/root/usr/share/luci/menu.d/luci-mod-status.json#L66
  https://github.com/openwrt/luci/blob/master/themes/luci-theme-openwrt/luasrc/view/themes/openwrt.org/header.htm#L203

feckert, 2020-01-21:
Please open a pull request with your suggested changes.

jow-, 2020-01-21:
I think its finally time to implement translation contexts: https://www.gnu.org/software/gettext/manual/html_node/Contexts.html

That would solve the problem neatly and it should be possible to introduce that without too much fuzz. We just need think about a suitable API. For brevity, I'd like to continue using `_()` instead of introducing another translation function name so maybe we can add a second optional context argument to it.

For example:

`_("Load", "Load remote data")` and `_("Load", "System load value")` 

which would then appear as

```
msgctxt "Load remote data"
msgid "Load"
msgstr "Laden..."

msgctxt "System load value"
msgid "Load"
msgstr "Systemlast"
```
in the PO files respectively. While we're at it we should probably also add plural translation support, e.g. `_#("One thing", "%d things", n);`

jow-, 2020-02-23:
Plural translations and context support is present in master now. One can use `_('Load', 'System load level')` to disambiguate *System Load* from *Data loading*.

closed 2020-02-23 by jow-