openwrt/luci #3506: vpn-policy-routing

leonardpitzu, 2020-01-15:
there is a recursive dependency which causes some error messages in the current master

`
tmp/.config-package.in:138935:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:138935:	symbol PACKAGE_vpn-policy-routing depends on PACKAGE_vpnbypass
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:138955:	symbol PACKAGE_vpnbypass depends on PACKAGE_vpn-policy-routing
`

jow-, 2020-01-16:
See #3507

closed 2020-01-16 by jow-

stangri, 2020-01-16:
Fixed in https://github.com/openwrt/packages/pull/11038