openwrt/luci #3528: Dropbear/SSH Settings missing

madduci, 2020-01-18:
I've updated my router to OpenWRT 19.07 stable, following also the [upgrade tutorial](https://openwrt.org/docs/guide-user/installation/ar71xx.to.ath79) to move from ar71xx to ath79.

Since the upgrade, I cannot see in LuCi the dropbear/ssh authentication settings anymore.

Additional information:

* Firmware Version: OpenWrt 19.07.0 r10860-a3ffeb413b / LuCI openwrt-19.07 branch git-20.016.32041-5baeb64
* Router Version: TP-Link Archer C7 v2.0

jow-, 2020-01-18:
Works for me. The settings are in System -> Administration. If you still can't find them, make sure that your dropbear configuration is not corrupted. Execute `uci show dropbear >/dev/null` on the command line to see potential errors.

closed 2020-01-18 by jow-

madduci, 2020-01-18:
In Systems->Administration i can only change the root password, but there's no dialog for Dropbear and I cannot login in my router to issue the command, since it is not configured/disabled.

Should I flash the factory image, erasing everything?

ysc3839, 2020-01-18:
It's here.
[image]
Also dropbear is enabled by default. If you can't login via SSH, I guess there's something wrong.

hnyman, 2020-01-18:
@jow- 
Apparently there is no menu tab, if there is no /etc/config/dropbear file at all.
I tested by renaming the file and the "SSH Access" tab is not shown.

[image]

madduci, 2020-01-18:
exactly, there's was no tab after the flashing of the sysupgrade firmware.

I have done a factory reset and now it's there, but the upgrade from 18.06.5 ar71xx to 19.07 ath9 had problems with configuration files.

hnyman, 2020-01-18:
Just for future reference: 
I figured out how it can be fixed also from LuCI...

* opkg install "luci-app-commands"  (Luci custom commands)
* create three commands that create the dropbear uci config file, enter there the simplest uci content, and commit it:
   * touch /etc/config/dropbear
   * uci add dropbear dropbear
   * uci commit dropbear
* run those custom commands from LuCI

That creates enough good dropbear config file that the LuCI config page gets shown properly.

[image]