openwrt/luci #3534: Can't set password in LuCI in Trunk

cvmiller, 2020-01-19:
Version:	OpenWrt SNAPSHOT r12040-8df14c229c / LuCI Master git-20.016.76613-0608ff0

Using LuCI on the Trunk (snapshot) with a fresh install on TP-LInk Archer C2 V1 AC750, unable to set the password from menu System->Administration, typing in the same password (revealing the password to check) get error:  "Failed to change the system password"

Can successfully set password via CLI/SSH with 'passwd' command.

jow-, 2020-01-19:
Already fixed with cbb1c3cf932e4f5d6d56b63ba2e87bd761bfcc54

Workaround:
```
root@OpenWrt:~# chmod +x /usr/libexec/rpcd/luci 
root@OpenWrt:~# /etc/init.d/rpcd restart
```

closed 2020-01-19 by jow-

thiscantbeserious, 2021-05-19:
Are you sure that's all? I'm on 21.02 and even did a new config to rule out any depenency issues like NGNIX/ssl then applied your above fix - did nothing for me.

I can set the password via 'ssh passwd' but afterwards the login fails on both LuCi and via SSH - passwd doesn't return any error. 

Maybe its related to anything in regards to encryption? Any idea how I could dive deeper into that issue?

cvmiller, 2021-05-19:
I downloaded a snapshot from 4 May and upgraded my router
OpenWrt SNAPSHOT, r16634-5d8ea6d34f

I was able to set the password via the LuCI web interface, and log in via SSH using that password. So it _appears_ to be working at least as recently as 4 May 2021.