openwrt/luci #3539: Problem with material theme

alisaeed, 2020-01-20:
Hello,
I found two problems in material theme :
1: Save&Apply drop down menu (both foreground and background are white):
[image]

2: Save button : 
[image]

jow-, 2020-01-20:
@LuttyYang

alisaeed, 2020-01-20:
We changed cascade.css in material theme as follow:

.cbi-dropdown[open] > ul.dropdown {
	position: absolute;
	z-index: 1100;
	display: block;
	width: auto;
	min-width: 100%;
	max-width: none;
	max-height: 200px !important;
	border: thin solid #918e8c;
	background: #f6f6f6;
	box-shadow: 0 0 4px #918e8c;
	**color: var(--main-menu-color);**
}

Result : 
[image]

feckert, 2020-01-20:
Thanks :+1:
fixed with your suggestion see commit https://github.com/openwrt/luci/commit/1c49bda198ef33c0423777a966e6c6ab477da34b will also cherry-pick this to the branch openwrt-19.07

feckert, 2020-01-20:
Also fixed with https://github.com/openwrt/luci/commit/c395d438074e598d6e51d5ed104f5712f79b63b6 in openwrt-19.07

closed 2020-01-20 by feckert