openwrt/luci #3540: Solved "Save" button UI problem in material theme

alisaeed, 2020-01-20:
Hello,
We found problem in material theme :

[image]

We Solved this problem in cascade.css as follow : 

.modal .spinning {
	`/*margin-bottom: 2em;*/`
}

feckert, 2020-01-21:
[image]

I can not reproduce your problem. What browser do you use?

alisaeed, 2020-01-21:
When you click save button this happens (save button changes to spinning button and this problem happens).

We Solved this problem in cascade.css as follow :
.modal .spinning {
/*margin-bottom: 2em;*/
}

alisaeed, 2020-01-21:
> [image]
> 
> I can not reproduce your problem. What browser do you use?

When you click save button this happens (save button changes to spinning button and this problem happens). I've tested this in different browsers.

feckert, 2020-01-21:
Ok thanks for the explanation. I could reproduce this.
Will remove this section from the css.
https://github.com/openwrt/luci/blob/cc01770fa1cf09b729dd931df77b149d1b20d2ef/themes/luci-theme-material/htdocs/luci-static/material/cascade.css#L1675-L1679

feckert, 2020-01-21:
Thanks for your fix.
 I have pushed this to master https://github.com/openwrt/luci/commit/71743a7ef69e9ffdf890bf895d704ecc0648548b.
@alisaeed Does we have to push this also to the openwrt-19.07 branch?

feckert, 2020-01-21:
Thanks for your work

closed 2020-01-21 by feckert