openwrt/luci #3542: luci-theme-material: small visual errors

castillofrancodamian, 2020-01-21:
**Giant buttons**

[image]
[image]

**Logout symbol attached to the text, not as in English that is separated**

[image]

**Misaligned buttons**

[image]

**Text attached to the table**

[image]

**And a lot of white space**

[image]
[image]
[image]
[image]

jow-, 2020-01-21:
@LuttyYang

castillofrancodamian, 2020-03-28:
I don't think it's going to respond. I see there is no activity on your GitHub profile.

dannil, 2024-02-27:
@stokito the only present issue seems to be "Text attached to the table", the rest is already fixed on the master branch. 

Regarding the white space in modals (= dialogs that gray out the background), I've assumed that's an intentional design decision of the theme since all modals is styled to always take up 90% of the width of the viewport. In the Bootstrap theme this isn't really noticeable since there the width is a fixed 900px. I assume that @LuttyYang is not active here anymore so getting an answer to that question will probably not happen, but my guess is that it's intentional since the Material theme is designed around the principle of taking up all available horizontal space.

stokito, 2024-02-27:
maybe then the issue can be closed, right?

dannil, 2024-02-27:
Yes, I would say so.

closed 2024-02-27 by castillofrancodamian