openwrt/luci #3546: [Bug] luci-app-wireguard: Too many peers results in background not rendering

Mushoz, 2020-01-21:
As the screenshot shows, when you have too many peers (4 in this case) the background stops rendering and things are extremely difficult to tell apart.

[image]

jow-, 2020-01-21:
Which LuCI version?

Mushoz, 2020-01-21:
I installed 19.07.0 on this device yesterday. These are the versions currently installed on the device:

luci-ssl-openssl	git-20.016.32041-5baeb64-1
luci	git-20.006.26738-35aa527-1

Functionality is unaffected by the way. All 4 peers work fine.

jow-, 2020-01-22:
Is the issue persisting after `opkg update; opkg upgrade luci-theme-bootstrap` and browser cache clearing?

Mushoz, 2020-01-22:
This fixes the issue, thank you! Closing :)

closed 2020-01-22 by Mushoz