openwrt/luci #3549: luci-app-openvpn: add "server-ipv6" option

Vantuz, 2020-01-21:
I want to set up similar config using uci/luci options

```
# general settings for openvpn server
local 192.0.2.197
proto udp
dev tun
# setting up an openvpn server with certificates is covered here: https://community.openvpn.net/openvpn/wiki/GettingStartedwithOVPN
ca myca.crt
cert myserver.crt
key myserver.key
dh dh1024.pem
# use 10.11.12.0/24 for the openvpn tunnel ipv4
server 10.11.12.0 255.255.255.0
# adittional settings for ipv6:
server-ipv6 2001:db8:f00:bebe::/64
push "route-ipv6 ::/0"
push "route-metric 2000"
```

Knogle, 2021-09-23:
Any news about that? I need it as well.

Neustradamus, 2021-10-20:
Yes and now, ISP gives only IPv6 addresses.

TobbeJ, 2024-12-22:
ran in to this exact issue, ipv6 related options is not available via luci so i got stuck getting ipv6 working over openvpn.
a bit surprised this is not picked up in 3 years, i assume using ipv6 over vpn is not something normally done.

systemcrash, 2024-12-22:
No more development is planned for LUA based apps, which where possible are being converted to JS. openvpn is a blocker since it seems to require conversion to a proto (in the main openwrt repo), and previous attempts staled. 

If you want to add the necessary parameters (in LUA), you're welcome.