openwrt/luci #3552: js: fs.write() to a non-existant path fails in hard to diagnose ways

karlp, 2020-01-23:
Using the following code in a js style CBI, to save a field to a custom file results in "save and apply" popping a dialog that "Cannot save due to invalid values" and a console error about "uncaught exception: undefined"

```
		o = s.taboption("blah", form.TextValue, "magic", _("test field"), _("just trying to save a file"));
		o.rows = 2;
		o.write = o.remove = function(sid, fv) {
			var fn = "/etc/doesnt-exist/anyfilename";
			if (typeof(fv) === "undefined") {
				console.log("removing pseudo file: ", fn);
				return fs.remove(fn);
			} else {
				console.log("Attempting to save pseudo file: ", fn);
				return fs.write(fn, fv.trim());
			}
		}
```

If the directory exists, everything behaves as expected.   It would be nice for this to propagate a better error.  Appears to only affect fs.write and fs.remove.  returning fs.trimmed() in a o.cfgvalue() call simply silently loads no data, which is perfectly reasonable behaviour when there's no file available to load.

jow-, 2020-01-29:
`fs.write()` returns a promise reject with error details in case of a failure. You need to catch that if your use-case requires a handling of the details. See e.g. https://github.com/openwrt/luci/blob/master/modules/luci-mod-system/htdocs/luci-static/resources/view/system/startup.js#L50 for an example.

closed 2020-01-29 by jow-

karlp, 2020-01-29:
ok, this is just "modern javascript" again then.  I'll adapt I guess :)

commit 2dfcca23 (2020-04-10): luci-base: form.js: improve save error handling

Fixes: #3552