openwrt/luci #3553: luci-base: "LED Configuration" is not available to translate

castillofrancodamian, 2020-01-23:
**Spanish**

[image]

**Portuguese**

[image]

**French**

[image]

hnyman, 2020-01-23:
Pure hunch, but the reason could be that this menu item has (for some strange reason) a abbreviation structure in the menu string.  That might cause some conflict.

The abbreviation does not even work in the menu item, so it should probably be removed from it.
(but it works in the page header of the .js page)

hnyman, 2020-01-23:
Likely fixed by 2a1a987

closed 2020-01-23 by hnyman

castillofrancodamian, 2020-01-23:
👍👍👍

manjuprajna, 2020-01-25:
@hnyman 
under luci zh-cn before 2a1a987, it's good, but after 2a1a987 become like this:
[image]
so need to do another translate for this?

hnyman, 2020-01-25:
The string needs to be re-translated into Chinese as there is now a new string for that menu item.

castillofrancodamian, 2020-01-25:
Ready!

[image]