openwrt/luci #3557: WOL Services not working using etherwake

wrtechnopath, 2020-01-24:
I've upgraded to OpenWrt 19.07.0 r10860-a3ffeb413b / LuCI openwrt-19.07 branch git-20.021.30444-421791d and wanted to utilize the wake-on-lan (WOL) function using LuCI. There are two available programs to execute this, wakeonlan and etherwake. LuCI => Services => Wake on LAN utilizes the etherwake program. It's a clean and simple UI but not functioning correctly.

I broadcast on all interfaces, to my web-server that I want to wake up (the MAC addresses and hostnames are already stored), I've tried also tried the broadcast address tick box with no luck but a success message.

jow-, 2020-01-24:
Does it work if you execute `wakeonlan` on the command line?

wrtechnopath, 2020-01-25:
Yes

jow-, 2020-01-25:
Which command line do you use exactly that works?

wrtechnopath, 2020-01-25:
wakeonlan -i 192.168.1.255 MA:CA:DD:RE:SS

hmh, 2020-01-25:
etherwake seems not to work if you tell it (through LuCI, did not try the CLI) to broadcast to all interfaces.  I use it in broadcast mode all the time from LuCI, but I *must* tell it to send on the correct interface (eth0.1 in my case), and to send to the broadcast address.

wrtechnopath, 2020-01-25:
Worked using the specific interface

hmh, 2020-01-26:
I am not sure this should be closed.  Either etherwake needs to be fixed to actually send to all interfaces when that is requested, or the LuCI interface should not show that option when etherwake is the backend...

wrtechnopath, 2020-01-26:
I agree, etherwake or luci-app-wol has a bug.

commit 40ee3fa3 (2020-01-26): luci-app-woll: convert to client side rendering

Also make the interface option mandatory

Fixes: #3557
closed 2020-01-26 by jow-