openwrt/luci #3565: Alice Gate AGPF - Infinite Spinning Whell when try to access "Network - Switch". How i can fix ?

DjDiabolik, 2020-01-26:
I have update OpenWrt... i have installed this from Luci:
https://openwrt.org/toh/pirelli/agpf

Apparently the flash it's be correct..... i have try to look around the new openwrt and i see when try to access to "Network - Switch" i obtain an infinite spinning wheel.
And it's also appears two errors whit a button say "Dismiss".
The errors say:
TypeError

portstate is null... it's colored RED. Dismiss button on right.

The second say:
Switch eth1 has an unknown topology - the VLAN settings might not be accurate. Colored Yellow and Dismiss button on right.

I have also tryed to click on both "Dismiss" button but apparently nothings append..... only the infinite spin remain and it's say "Loading view…" but nothings appears also after 10 minutes.

How i can fix this ?

jow-, 2020-01-26:
What is reported by the following two shell commands?

```
ubus call luci getSwconfigFeatures '{ "switch": "eth1" }'
ubus call luci getSwconfigPortState '{ "switch": "eth1" }'
```

DjDiabolik, 2020-01-26:
The first it's:
[code]
root@OpenWrt:~# ubus call luci getSwconfigFeatures '{ "switch": "eth1" }'
{
        "num_vlans": 16,
        "min_vid": 1,
        "vlan_option": "enable_vlan",
        "switch_title": "BCM5325"
}
root@OpenWrt:~#
[/code]
The second:
[code]
root@OpenWrt:~# ubus call luci getSwconfigPortState '{ "switch": "eth1" }'
{
        "result": [
                {
                        "duplex": true,
                        "port": 0,
                        "txflow": false,
                        "link": true,
                        "speed": 100,
                        "auto": true,
                        "rxflow": false
                },
                {
                        "duplex": false,
                        "port": 1,
                        "txflow": false,
                        "link": false,
                        "speed": 0,
                        "auto": false,
                        "rxflow": false
                },
                {
                        "duplex": true,
                        "port": 2,
                        "txflow": false,
                        "link": true,
                        "speed": 100,
                        "auto": true,
                        "rxflow": false
                },
                {
                        "duplex": true,
                        "port": 3,
                        "txflow": false,
                        "link": true,
                        "speed": 100,
                        "auto": true,
                        "rxflow": false
                },
                {
                        "duplex": false,
                        "port": 4,
                        "txflow": false,
                        "link": false,
                        "speed": 0,
                        "auto": false,
                        "rxflow": false
                },
                {
                        "duplex": true,
                        "port": 5,
                        "txflow": false,
                        "link": true,
                        "speed": 100,
                        "auto": false,
                        "rxflow": false
                }
        ]
}
root@OpenWrt:~#
[/code]

DjDiabolik, 2020-01-28:
sorry @jow- but i need some help (how i need to do somethings).
You can say if the shell output it's ok or it's need to check somethings on my current setup ?
I have also a open thread on forum (in hope someone can help.. someone maybe non regged here on github) but 0 reply at moment........

jow-, 2020-01-28:
Please provide me with the output of `swconfig dev eth1 show` as well.

jow-, 2020-01-28:
Fixed by 616d44c155ae2b4f19e8b2645744a107e38a56c7 and 6d59a6400ed055d71e0b335679d291c22bbdbd40

closed 2020-01-28 by jow-

DjDiabolik, 2020-01-28:
@jow-  ok... i do it.... first:
``
root@OpenWrt:~# swconfig dev eth1 show
Global attributes:
        enable_vlan: 1
        ports: 0x003f
Port 0:
        pvid: 1
        link: port:0 link:up speed:100baseT full-duplex auto
Port 1:
        pvid: 1
        link: port:1 link:up speed:100baseT full-duplex auto
Port 2:
        pvid: 1
        link: port:2 link:up speed:100baseT full-duplex auto
Port 3:
        pvid: 1
        link: port:3 link:up speed:100baseT full-duplex auto
Port 4:
        pvid: 1
        link: port:4 link:down
Port 5:
        pvid: 1
        link: port:5 link:up speed:100baseT full-duplex
VLAN 1:
        ports: 0 1 2 3 4 5
root@OpenWrt:~#
``

The seconds:
``
root@OpenWrt:~# cat /etc/board.json
{
        "model": {
                "id": "agpf-s0",
                "name": "Pirelli Alice Gate AGPF-S0"
        },
        "network": {
                "lan": {
                        "ifname": "eth1",
                        "protocol": "static"
                },
                "wan": {
                        "ifname": "eth0",
                        "protocol": "dhcp"
                }
        },
        "switch": {
                "switch0": {
                        "enable": true,
                        "reset": true,
                        "ports": [
                                {
                                        "num": 0,
                                        "role": "lan",
                                        "index": 4
                                },
                                {
                                        "num": 1,
                                        "role": "lan",
                                        "index": 3
                                },
                                {
                                        "num": 2,
                                        "role": "lan",
                                        "index": 2
                                },
                                {
                                        "num": 3,
                                        "role": "lan",
                                        "index": 1
                                },
                                {
                                        "num": 5,
                                        "device": "eth1",
                                        "need_tag": false,
                                        "want_untag": true
                                }
                        ],
                        "roles": [
                                {
                                        "role": "lan",
                                        "ports": "0 1 2 3 5",
                                        "device": "eth1"
                                }
                        ]
                }
        }
}
``

DjDiabolik, 2020-01-29:
@jow- i know i'm noob :)
I have right now update all my package from "software" section..... apparently all it's be updated corrected also the package whit this recent commit "luci-mod-network".

After i have update all avaible package i have perform a reboot of my AGPF.... tryed to access again to "Network -> Switch" but apparently it's not resolve the issue.
I continue to obtain the two errors and the infinite spin "Loading view".

How i have read some thread in forum about similar issue and i see there's some people report a problems when using "Firefox" "Chrome" or other "Chronium" based browser.
In my case i have tryed to access to this "Switch" also from Edge....... apparently there's no help to resolve this issue..

jow-, 2020-01-29:
You need to wait a bit before the updated packages become available. It also depends on your installed version. The fix for 19.07 for example was pushed just half an hour ago. You also might need to clear your browser cache to pick up the new files.

DjDiabolik, 2020-01-30:
@jow- oh.. interesting.
Also from right now when i click on "Update list" i obtain an errors say:
```
Collected errors:
 * pkg_hash_add_from_file: Failed to open /var/opkg-lists/openwrt_routing: Out of memory.
```

I need to clear browser cache ?...... damn.... i lost al cookie password and login for all my forums and site :(

DjDiabolik, 2020-01-30:
Apparently a reboot as resolve... i have correctly obtain the new package list.
For a specific upgrade it's say:

```
Upgrading luci-base on root from git-20.026.75896-f519510-1 to git-20.029.31300-19a48f6-1...
Downloading http://downloads.openwrt.org/releases/19.07.0/packages/mips_mips32/luci/luci-base_git-20.029.31300-19a48f6-1_mips_mips32.ipk
Configuring luci-base.
```

```
Collected errors:
 * resolve_conffiles: Existing conffile /etc/config/luci is different from the conffile in the new package. The new conffile will be placed at /etc/config/luci-opkg.
 * resolve_conffiles: Existing conffile /etc/config/ucitrack is different from the conffile in the new package. The new conffile will be placed at /etc/config/ucitrack-opkg.

```

It's normal ?  I have to rename after the config file manually ?

DjDiabolik, 2020-01-30:
@jow- omg... now switch are correctly opened:
I continue to see the errors about "Switch1 unknow topology"....

Sample screenshot how it's not the page:
https://imgur.com/a/QpfBQed

What's the the Seconds line ? It's all say untagged...... and the "1" it's normal ? I can resolve here the issue about "uknow topology" ? I can delete the seconds line ?

jow-, 2020-01-30:
I don't know if the switch settings are correct for your particular board. The unknown topology error means that LuCI cannot identify the hardware layout correctly. What is the contents of `/etc/board.json` on your router?

DjDiabolik, 2020-01-30:
@jow- simply ask... if it's possible to resolve also this issue i can report to you anythings:
```
{
	"model": {
		"id": "agpf-s0",
		"name": "Pirelli Alice Gate AGPF-S0"
	},
	"network": {
		"lan": {
			"ifname": "eth1",
			"protocol": "static"
		},
		"wan": {
			"ifname": "eth0",
			"protocol": "dhcp"
		}
	},
	"switch": {
		"switch0": {
			"enable": true,
			"reset": true,
			"ports": [
				{
					"num": 0,
					"role": "lan",
					"index": 4
				},
				{
					"num": 1,
					"role": "lan",
					"index": 3
				},
				{
					"num": 2,
					"role": "lan",
					"index": 2
				},
				{
					"num": 3,
					"role": "lan",
					"index": 1
				},
				{
					"num": 5,
					"device": "eth1",
					"need_tag": false,
					"want_untag": true
				}
			],
			"roles": [
				{
					"role": "lan",
					"ports": "0 1 2 3 5",
					"device": "eth1"
				}
			]
		}
	}
}
```

Otherwise i know my router as obtain his last image avaible and openwrt supports it's at end of life for this device.

A stupid question:
I have kept the previous configuration file when i flash this new version of openwrt.... it may be this that created this issue/trouble ?

DjDiabolik, 2020-02-01:
Hey @jow- I have a GREAT NEWS for you!
From the openwrt home page i see it's release the 19.07.1 also for my device.

I have download the generic image and i have flashed on my device.
After upload complete i have verified the SHA256 and i have CHOOSE to not maintain the current configuration!!!

After some minutes i have need to reconfigure all from beginning...... i have to do:
- Connect my PC directly to this device
- From the 192.168.1.1 revert to use 192.168.1.254
- I have reconfigure/reenable the wifi
- I have disabled the DHCP for ipv4 
- If i do this correct i have also disabled the ipv6
- And some other preferences.

After that........ YEAHH! Now the Network -> SWITCH It's be opening normally!!
Not unknow topology or other things......... If you want ask and i can upload here the content of currently file so that we can deal with those that didn't work.

Previously for fix all situation problably i can reset to default the configuration..... now i'm leave all how it's released whitout updating any package instead problably it's not need in this "old" device.