openwrt/luci #3567: "uci: Entry not found" after upgrade to OpenWrt 19.07.0

richb-hanover, 2020-01-26:
I just upgraded to OpenWrt 19.07.0 and was restoring my configuration, and was surprised to see a "uci: Entry not found" in a script that had previously worked flawlessly.

So I rebooted, and the first command I gave was `uci commit` and it got the same error. Is this expected? Here's the terminal session... 

Many thanks!

```
OpenWrt 19.07.0, r10860-a3ffeb413b
 -----------------------------------------------------
root@Main-Router:~#
root@Main-Router:~# uci commit
uci: Entry not found
root@Main-Router:~# uptime
 10:51:38 up 1 min,  load average: 1.62, 0.62, 0.22
root@Main-Router:~#
```

jow-, 2020-01-26:
Check your `/etc/config/` for junk files.

richb-hanover, 2020-01-26:
Ahah! That's the answer. 

I used the script described at https://forum.openwrt.org/t/script-to-list-installed-packages-for-simplifying-sysupgrade/7188/7 to preserve the list of installed packages (and then re-install after sysupgrade). The script saves the list in in `/etc/config/` 

Deleting `/etc/config/opkg.installed` solves the "uci: Entry not found" message.

closed 2020-01-26 by richb-hanover