openwrt/luci #3573: LuCI: sysauth is behaving weirdly on legacy controllers

hmh, 2020-01-27:
With legacy contollers (no JSON):

Suppose you have nodes:
* **/admin/branch** (type firstchild())
* **/admin/branch/a** (type template())
* **/admin/branch/b** (type template())

If one never touches the "_sysauth_" field of any of the nodes above, all works well and authentication is required to access any of them.

If one sets _sysauth = false_ on node **/admin/branch/a**, all works as expected: one can access **/admin/branch/a** while unauthenticated, and if one tries to access **/admin/branch**, one is forwarded to **/admin/branch/a** _and_ it works unauthenticated as it should.  And any attempts to access **/admin/branch/b** require authentication, as it should.

However, if one sets the sysauth field of **/admin/branch** to false, it disables authentication requirements for both **/admin/branch/a** and **/admin/branch/b**.

On a related note, setting sysauth = true on a node breaks authentication for that node.  I am not entirely sure this was an API break with legacy LuCI, or if setting sysauth = true has always been a wrong thing to do.

hmh, 2020-01-27:
Note: when I said "no JSON", I meant in the controller that sets up **/admin/branch**, the LuCI upstream stuff that is JSON based is there unmodified and active (e.g. luci-admin-full, luci-base)...

hmh, 2020-01-27:
Hmm, this behavior (sysauth=false overrides sysauth on tree children) is also present on 18.06, so it is not a regression.

jow-, 2020-01-28:
This is by design, I have no plans to change this. If you feel it should be improved I would take patches changing the behavior.

closed 2020-01-28 by jow-

hmh, 2020-01-28:
Thanks for the quick reply!  Knowing that this is by design does help, although I wonder about the worse behavior in 19.07 and earlier (re. firstchild() and firstnode() redirections requiring auth before doing the redirection), but since that behavior is very security-sensitive, it seems unwise to patch it to behave like the new JSON-based code.

As for providing patches, I do try.  But the LuCI developer documentation is so sparse, especially the internals and data structures, that trying to change anything more fundamental (or even trying to track down a possible bug), required non-trivial amounts of reverse engineering effort.  It has proven to be quite an effective barrier to contributing code and patches :-(

Please consider this a plea to take the time to document all data structures and their fields, and the overall flow and how things are tied together during the ongoing refactoring.  Only someone that deeply understands the code (and the history of the data structures involved since there is 19.07 LuCI and earlier to consider when documenting LuCI behavior) can do that.

Also, please consider refusing to merge patches from third parties that change or add a data structure or public API to LuCI until the patch author adds the appropriate documentation.