openwrt/luci #3576: Interface stat bug in 18.06

y-alexey, 2020-01-27:
Display of interface packets number by negative number.
[image]
`eth0.56   Link encap:Ethernet  HWaddr 28:2C:B2:EE:76:D5  
          inet addr:10.102.112.38  Bcast:10.102.112.255  Mask:255.255.255.0
          inet6 addr: fe80::2a2c:b2ff:feee:76d5/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1293143934 errors:0 dropped:585822 overruns:0 frame:0
          TX packets:2697179389 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:273981505947 (255.1 GiB)  TX bytes:2362350068498 (2.1 TiB)` actually.
Mipsel_24kc traget, if it's matter.

feckert, 2020-01-29:
So far, no one has noticed. So I guess it's not a serious bug. I would like to note that 18.06 is slowly moving EOL. Can you check with the new stable 19.07 to see if we have the same problem. The LuCI has changed from 19.07 so that not everywhere Lua but Javascript is used. The network configuration page is completely new written in Javascript.

y-alexey, 2020-01-29:
This, moreover, is a lua-mod according to the instructions from the wiki.

For me, such a display is not important.  And the router, as you might guess, works in production.  So it’s unlikely that I’m going to update it to 19.07 soon.

But, of course, I will pay attention and update the issue.

jow-, 2020-01-29:
Can you please provide the output of `ubus call network.device status` ?

jow-, 2020-01-29:
I believe this issue is the same as #3003 and it should have been fixed by the backport in 41e2258d6dc1ebe8d3874ae6d6b13db49cff2c5c

closed 2020-01-29 by jow-