openwrt/luci #3583: luci-mod-system: ZRam settings dopdown has dublicated values

stonerl, 2020-01-28:
The ZRam settings have a dropdown for the compression algorithm which is prefilled with `lzo`. The problem is that `lzo`is also available as a possible value.

[image]

This makes it confusing because it is not obvious that the first value is only a placeholder. That also does not disappear when a proper value is selected.

https://github.com/openwrt/luci/blob/6d59a6400ed055d71e0b335679d291c22bbdbd40/modules/luci-mod-system/htdocs/luci-static/resources/view/system/system.js#L190

feckert, 2020-01-29:
Could you please rename this line to `o.default = 'lzo';`?
Maybe this will fix your issue. If so could you pleas report this back so we could fix this in the master branch.

closed 2020-01-29 by jow-

stonerl, 2020-01-29:
Thank you for the quick fix.

commit b8f65c34 (2020-01-29): luci-mod-system: fix zram compression placeholder

Fixes: #3583
commit 3f5a67b6 (2020-01-29): luci-mod-system: fix zram compression placeholder

Fixes: #3583

(cherry picked from commit b8f65c340c42acda920298af4318784a79a0965c)