openwrt/luci #3584: WPS box/option missing in Luci (19.07.0)

CaboBeach, 2020-01-28:
Doubt very many use this, so not very important, but i noticed after installing wpad full and hostapd-utils that the WPS box/option was missing in Network>Wireless>Wireless Security in 19.07.0, looked in other sections but couldn't find it anywhere.

[image]

castillofrancodamian, 2020-01-28:
Did you try restarting the router?

CaboBeach, 2020-01-28:
Yes, but now that i think about it i haven't tried resetting or reflashing, so i'll give that a try in the next couple of days.

jow-, 2020-01-29:
You need to close and reopen your browser tab after installing these packages.

closed 2020-01-29 by jow-

CaboBeach, 2020-01-31:
Just wanted to mention that the wps box/option is there in 19.07.1.