openwrt/luci #3586: luci-proto-modemmanager: use of /dev/cdc-wdm0

Simcra, 2020-01-29:
So I have been able to create and install libqmi, libmbim and modemmanager packages successfully, however, I am having difficulty with the luci-proto-modemmanager package.

When I select the modem device using the luci module, it fails to connect.

However if I execute `mmcli -L` to list the modems, and then `mmcli -m 0` where 0 is the modem identifier, I can obtain the system device path: '/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4'

Updating the device path in /etc/config/network manually to the system device path and rebooting somehow enables the modem to connect without error.

Older versions of the luci-proto-modemmanager, prior to the merge into the official openwrt git use the system device path, I have tried to modify the luci-proto-modemmanager locally to execute mmcli -m 0  and filter out the device path for the dropdowns to no avail.

Software versions:
- OpenWRT 19.07.0
- libmbim 1.20.4
- libqmi 1.24.4
- modemmanager 1.12.4
- luci-proto-modemmanager latest from master

feckert, 2020-01-29:
From my point of view this is not supported by 19.07 because the dependency were not satisfied at release.
So we removed the luci-proto-modemmanger from the 19.07 branch https://github.com/openwrt/luci/commit/a13689c800f536d2c19d0b31c45fed0a4b255eda

Yes I have seen that we now have the dependency for this branch now satisfied.
https://github.com/openwrt/packages/pull/11149

I always thought that **no** new packages would be added, only bug fixes.

@jow- What should we do now? Do a revert from a revert of https://github.com/openwrt/luci/commit/a13689c800f536d2c19d0b31c45fed0a4b255eda ?

jow-, 2020-01-29:
@feckert I don't think this ticket is about 19.07. but about the device dropdown suggesting `/dev/whatever` instead of `/sys/devices/blah`. Probably needs a change in the backend ubus procedure to resolve the `/dev/whatever` symlinks to their `/sys/devices/blah` targets. Due to a lack of understanding, hardware, time and interest I can't take care of this though.

jow-, 2020-01-29:
@Simcra - it would be helpful if you could provide the output of `ls -lh /dev`.

Simcra, 2020-01-29:
root@BT3001:~# ls -lh /dev
drwxr-xr-x    3 root     root          60 Jan  1  1970 bus
crw-------    1 root     root      180, 176 Jan 29 17:56 cdc-wdm0
crw-------    1 root     root        5,   1 Jan  1  1970 console
crw-------    1 root     root       10,  63 Jan  1  1970 cpu_dma_latency
crw-------    1 root     root       29,   0 Jan  1  1970 fb0
crw-rw-rw-    1 root     root        1,   7 Jan  1  1970 full
crw-------    1 root     root      254,   0 Jan  1  1970 gpiochip0
crw-------    1 root     root      254,   1 Jan  1  1970 gpiochip1
crw-------    1 root     root      254,   2 Jan  1  1970 gpiochip2
crw-------    1 root     root      254,   3 Jan 29 17:56 gpiochip3
crw-------    1 root     root      254,   4 Jan 29 17:56 gpiochip4
crw-------    1 root     root       89,   1 Jan 29 17:56 i2c-1
drwxr-xr-x    2 root     root          60 Jan  1  1970 input
crw-------    1 root     root        1,  11 Jan  1  1970 kmsg
srw-rw-rw-    1 root     root           0 Jan 29 17:56 log
crw-------    1 root     root       10, 237 Jan  1  1970 loop-control
brw-------    1 root     root        7,   0 Jan  1  1970 loop0
brw-------    1 root     root        7,   1 Jan  1  1970 loop1
brw-------    1 root     root        7,   2 Jan  1  1970 loop2
brw-------    1 root     root        7,   3 Jan  1  1970 loop3
brw-------    1 root     root        7,   4 Jan  1  1970 loop4
brw-------    1 root     root        7,   5 Jan  1  1970 loop5
brw-------    1 root     root        7,   6 Jan  1  1970 loop6
brw-------    1 root     root        7,   7 Jan  1  1970 loop7
crw-------    1 root     root        1,   1 Jan  1  1970 mem
crw-------    1 root     root       10,  60 Jan  1  1970 memory_bandwidth
brw-------    1 root     root      179,   0 Jan  1  1970 mmcblk0
brw-------    1 root     root      179,   1 Jan  1  1970 mmcblk0p1
brw-------    1 root     root      179,   2 Jan  1  1970 mmcblk0p2
crw-------    1 root     root       10,  62 Jan  1  1970 network_latency
crw-------    1 root     root       10,  61 Jan  1  1970 network_throughput
crw-rw-rw-    1 root     root        1,   3 Jan  1  1970 null
crw-------    1 root     root      108,   0 Jan 29 17:56 ppp
crw-rw-rw-    1 root     root        5,   2 Jan 30 08:50 ptmx
drwxr-xr-x    2 root     root           0 Jan  1  1970 pts
brw-------    1 root     root        1,   0 Jan  1  1970 ram0
brw-------    1 root     root        1,   1 Jan  1  1970 ram1
brw-------    1 root     root        1,  10 Jan  1  1970 ram10
brw-------    1 root     root        1,  11 Jan  1  1970 ram11
brw-------    1 root     root        1,  12 Jan  1  1970 ram12
brw-------    1 root     root        1,  13 Jan  1  1970 ram13
brw-------    1 root     root        1,  14 Jan  1  1970 ram14
brw-------    1 root     root        1,  15 Jan  1  1970 ram15
brw-------    1 root     root        1,   2 Jan  1  1970 ram2
brw-------    1 root     root        1,   3 Jan  1  1970 ram3
brw-------    1 root     root        1,   4 Jan  1  1970 ram4
brw-------    1 root     root        1,   5 Jan  1  1970 ram5
brw-------    1 root     root        1,   6 Jan  1  1970 ram6
brw-------    1 root     root        1,   7 Jan  1  1970 ram7
brw-------    1 root     root        1,   8 Jan  1  1970 ram8
brw-------    1 root     root        1,   9 Jan  1  1970 ram9
crw-rw-rw-    1 root     root        1,   8 Jan  1  1970 random
drwxr-xr-x    2 root     root          60 Jan  1  1970 raw
crw-------    1 root     root      253,   0 Jan 29 17:56 rtc0
lrwxrwxrwx    1 root     root           8 Jan  1  1970 shm -> /tmp/shm
drwxr-xr-x    2 root     root          60 Jan 29 17:56 snd
crw-rw-rw-    1 root     root        5,   0 Jan  1  1970 tty
crw-rw----    1 root     dialout     4,   0 Jan  1  1970 tty0
crw-rw----    1 root     dialout     4,   1 Jan  1  1970 tty1
crw-rw----    1 root     dialout     4,  10 Jan  1  1970 tty10
crw-rw----    1 root     dialout     4,  11 Jan  1  1970 tty11
crw-rw----    1 root     dialout     4,  12 Jan  1  1970 tty12
crw-rw----    1 root     dialout     4,  13 Jan  1  1970 tty13
crw-rw----    1 root     dialout     4,  14 Jan  1  1970 tty14
crw-rw----    1 root     dialout     4,  15 Jan  1  1970 tty15
crw-rw----    1 root     dialout     4,  16 Jan  1  1970 tty16
crw-rw----    1 root     dialout     4,  17 Jan  1  1970 tty17
crw-rw----    1 root     dialout     4,  18 Jan  1  1970 tty18
crw-rw----    1 root     dialout     4,  19 Jan  1  1970 tty19
[... 82 lines cut ...]
  Hardware |      manufacturer: Generic
           |             model: MBIM [8087:0911]
           | firmware revision: XMM71XX_MAINT_V1.2_MBIM_TELSTR
           |      h/w revision: XMM7160_V1.2_MBIM_NAND_TELSTRA
           |         supported: gsm-umts, lte
           |           current: gsm-umts, lte
           |      equipment id: 354940080791469
  -----------------------------
  System   |            device: /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4
           |           drivers: cdc_mbim, cdc_acm
           |            plugin: Generic
           |      primary port: cdc-wdm0
           |             ports: cdc-wdm0 (mbim), ttyACM0 (at), wwan0 (net), ttyACM2 (at)
  -----------------------------
  Status   |             state: failed
           |     failed reason: sim-missing
           |       power state: on
           |    signal quality: 0% (cached)
  -----------------------------
  Modes    |         supported: allowed: 2g, 3g, 4g; preferred: none
           |           current: allowed: any; preferred: none
  -----------------------------
  IP       |         supported: ipv4, ipv6, ipv4v6

Hope this helps.

Simcra, 2020-01-29:
> @feckert I don't think this ticket is about 19.07. but about the device dropdown suggesting `/dev/whatever` instead of `/sys/devices/blah`. Probably needs a change in the backend ubus procedure to [...]

Yeah, I've tried doing this myself but I don't understand how the filters work and callFileExec from javascript. I've got modemmanager, libqmi and libmbim working correctly in 19.07.0, just need to work out how to resolve the symlinks.

jow-, 2020-01-30:
It seems your `/dev/cdc-wdm0` is not a symlink but a real char device, so no idea how to get to the other value there. What are the exact outputs of `mmcli -L` and `mmcli -m 0` ?

Simcra, 2020-01-30:
root@OpenWrt:~# mmcli -L
    /org/freedesktop/ModemManager1/Modem/0 [Generic] MBIM [8087:0911]

root@OpenWrt:~# mmcli -m 0
  -----------------------------
  General  |         dbus path: /org/freedesktop/ModemManager1/Modem/0
           |         device id: 4f1829603b75dd6e14b8b039e1b4bf313640c60f
  -----------------------------
  Hardware |      manufacturer: Generic
           |             model: MBIM [8087:0911]
           | firmware revision: XMM71XX_MAINT_V1.2_MBIM_TELSTR
           |      h/w revision: XMM7160_V1.2_MBIM_NAND_TELSTRA
           |         supported: gsm-umts, lte
           |           current: gsm-umts, lte
           |      equipment id: 354940080791469
  -----------------------------
  System   |            device: /sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.4
           |           drivers: cdc_mbim, cdc_acm
           |            plugin: Generic
           |      primary port: cdc-wdm0
           |             ports: cdc-wdm0 (mbim), ttyACM0 (at), wwan0 (net), ttyACM2 (at)
  -----------------------------
  Status   |             state: failed
           |     failed reason: sim-missing
           |       power state: on
           |    signal quality: 0% (cached)
  -----------------------------
  Modes    |         supported: allowed: 2g, 3g, 4g; preferred: none
           |           current: allowed: any; preferred: none
  -----------------------------
  IP       |         supported: ipv4, ipv6, ipv4v6

Please note the SIM card is not inserted right now

jow-, 2020-01-31:
Ugh... that code really wasn't written with machine readability in mind.

Please try the following untested patch:
```diff
diff --git a/protocols/luci-proto-modemmanager/htdocs/luci-static/resources/protocol/modemmanager.js b/protocols/luci-proto-modemmanager/htdocs/luci-static/resources/protocol/modemmanager.js
index 68dfb688c..804c567fa 100644
--- a/protocols/luci-proto-modemmanager/htdocs/luci-static/resources/protocol/modemmanager.js
+++ b/protocols/luci-proto-modemmanager/htdocs/luci-static/resources/protocol/modemmanager.js
@@ -1,21 +1,40 @@
 'use strict';
-'require rpc';
+'require fs';
 'require form';
 'require network';
 
-var callFileList = rpc.declare({
-	object: 'file',
-	method: 'list',
-	params: [ 'path' ],
-	expect: { entries: [] },
-	filter: function(list, params) {
-		var rv = [];
-		for (var i = 0; i < list.length; i++)
-			if (list[i].name.match(/^cdc-wdm/))
-				rv.push(params.path + list[i].name);
-		return rv.sort();
-	}
-});
+function getModemList() {
+	return fs.exec_direct('/usr/bin/mmcli', [ '-L' ]).then(function(res) {
+		var lines = (res || '').split(/\n/),
+		    tasks = [];
+
+		for (var i = 0; i < lines.length; i++) {
+			var m = lines[i].match(/\/Modem\/(\d+)/);
+			if (m)
+				tasks.push(fs.exec_direct('/usr/bin/mmcli', [ '-m', m[1] ]));
+		}
+
+		return Promise.all(tasks).then(function(res) {
+			var modems = [];
+
+			for (var i = 0; i < res.length; i++) {
+				var man = res[i].match(/manufacturer: ([^\n]+)/),
+				    mod = res[i].match(/model: ([^\n]+)/),
+				    dev = res[i].match(/device: ([^\n]+)/);
+
+				if (dev) {
+					modems.push({
+						device:       dev[1].trim(),
+						manufacturer: (man ? man[1].trim() : '') || '?',
+						model:        (mod ? mod[1].trim() : '') || dev[1].trim()
+					});
+				}
+			}
+
+			return modems;
+		});
+	});
+}
 
 network.registerPatternVirtual(/^mobiledata-.+$/);
 network.registerErrorCode('CALL_FAILED', _('Call failed'));
@@ -57,9 +76,10 @@ return network.registerProtocol('modemmanager', {
 		o = s.taboption('general', form.ListValue, 'device', _('Modem device'));
 		o.rmempty = false;
 		o.load = function(section_id) {
-			return callFileList('/dev/').then(L.bind(function(devices) {
+			return getModemList().then(L.bind(function(devices) {
 				for (var i = 0; i < devices.length; i++)
-					this.value(devices[i]);
+					this.value(devices[i].device,
+						'%s - %s'.format(devices[i].manufacturer, devices[i].model));
 				return form.Value.prototype.load.apply(this, [section_id]);
 			}, this));
 		};
diff --git a/protocols/luci-proto-modemmanager/usr/share/rpcd/acl.d/luci-proto-modemmanager.json b/protocols/luci-proto-modemmanager/usr/share/rpcd/acl.d/luci-proto-modemmanager.json
new file mode 100644
index 000000000..716f4c465
--- /dev/null
+++ b/protocols/luci-proto-modemmanager/usr/share/rpcd/acl.d/luci-proto-modemmanager.json
@@ -0,0 +1,12 @@
+{
+	"luci-proto-modemmanager": {
+		"description": "Grant access to mmcli",
+		"read": {
+			"cgi-io": [ "exec" ],
+			"file": {
+				"/usr/bin/mmcli -L": [ "exec" ],
+				"/usr/bin/mmcli -m [0-9]": [ "exec" ]
+			}
+		}
+	}
+}
```

I've also uploaded the individual files here: https://gist.github.com/jow-/d75fa2fb95e7b62736befc5c0398857d

Place the [luci-proto-modemmanager.json](https://gist.githubusercontent.com/jow-/d75fa2fb95e7b62736befc5c0398857d/raw/8a418e9cd1396c9c4831c422e84898a7a43e224a/luci-proto-modemmanager.json) file in `/usr/share/rpcd/acl.d/luci-proto-modemmanager.json` and the [modemmanager.js](https://gist.githubusercontent.com/jow-/d75fa2fb95e7b62736befc5c0398857d/raw/8a418e9cd1396c9c4831c422e84898a7a43e224a/modemmanager.js) in `/var/www/luci-static/resources/protocol/modemmanager.js` on your router. Log out and in again in LuCI and clear your browser cache.

feckert, 2020-01-31:
@jow as i can see in the source of mmcli that this tool supports a json output.
https://github.com/openwrt/packages/blob/master/net/modemmanager/Makefile#L11
This is the version we have in openwrt now (at least in master).
https://gitlab.freedesktop.org/mobile-broadband/ModemManager/commits/1.12.4
This commits adds the json output.
https://gitlab.freedesktop.org/mobile-broadband/ModemManager/commit/d1ac5ca16bfbc2475eb379b48e981e035f908b50

Simcra, 2020-02-04:
This patch does not work in 19.07.0, as the exec_direct command isn't defined in fs.js, I am in the process of updating OpenWRT to 19.07.1 for my target, I will let you know the results.

commit 58c091ac (2020-02-08): luci-proto-modemmanager: use nmcli to detemrine modem choices

Fixes: #3586
closed 2020-02-08 by jow-