openwrt/luci #3587: luci-app-advanced-reboot on LinksysWRT1900AC

jhthomas, 2020-01-29:
Hi,

luci-app-advanced-reboot to the alternate partition does not appear to be working on a LinksysWRT1900AC.

I am running19.07.0  firmware with nearly the stock setup of the router with the exception that I'm running Stubby and the Base IP is 10.0.0.2 rather than 192.168.1.1

LuCI openwrt-19.07 branch (git-20.024.68091-0bef180) / OpenWrt 19.07.0 r10860-a3ffeb413b
luci-app-advanced-reboot     git-20.028.17737-038050a-51

Model parameters listed as follows:

root@LinksysWRT1900AC:~# source /usr/share/libubox/jshn.sh; json_load "$(/bin/ubus call system board)"; json_get_vars model; echo $model
Linksys WRT1900AC
root@LinksysWRT1900AC:~# cat /tmp/sysinfo/board_name
linksys,mamba

Reboot to current partition from Advanced Tab  works as expected.

Attempt to boot to alternate partition yields an error:

Failed to execute call dispatcher target for entry '/admin/system/advanced_reboot/alternative_reboot'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/advanced_reboot.lua:258: bad argument #1 to 'new' (string expected, got table)
stack traceback:
	[C]: in function 'new'
	/usr/lib/lua/luci/controller/advanced_reboot.lua:258: in function </usr/lib/lua/luci/controller/advanced_reboot.lua:191>
	(tail call): ?

Attempt to use the Advanced Tab to Power Off yields:

Failed to execute call dispatcher target for entry '/admin/system/advanced_reboot/power_off'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/advanced_reboot.lua:291: bad argument #1 to 'new' (string expected, got table)
stack traceback:
	[C]: in function 'new'
	/usr/lib/lua/luci/controller/advanced_reboot.lua:291: in function </usr/lib/lua/luci/controller/advanced_reboot.lua:275>
	(tail call): ?

Booting to the alternate partition works fine from the command line using the following commands:

> fw_printenv boot_part   
> 1

> fw_setenv boot_part 2 

Booting to the alternate partition by toggling the power three times also works.

I would be delighted to hear your thoughts.

jow-, 2020-01-29:
@stangri

closed 2020-01-29 by jow-

jhthomas, 2020-01-30:
Hi Jo,

Thanks for your quick work.   2 of 3 functions are working as I would expect.   Reboot to the Alternate partition works.   Power Off also works ... or at least it shuts down the CPU but leaves the radios running.  This is probably the expected behavior for this hardware because there is only a hard-power switch for the router and no soft-power switch that I am aware of.

However, the Reboot to current partition does not work.   It throws essentially the same error  as shown in the original bug reports for the Reboot to Alternate and Power Off.   To be precise, the Reboot to Current partition button throws the following error:

Failed to execute call dispatcher target for entry '/admin/system/advanced_reboot/reboot'.
The called action terminated with an exception:
/usr/lib/lua/luci/controller/advanced_reboot.lua:186: bad argument #1 to 'new' (string expected, got function)
stack traceback:
	[C]: in function 'new'
	/usr/lib/lua/luci/controller/advanced_reboot.lua:186: in function </usr/lib/lua/luci/controller/advanced_reboot.lua:182>
	(tail call): ?

Finally, the Reboot function on the non-Advanced tab works as expected.

jow-, 2020-01-30:
Did you apply those fixes manually? From reading the code I don't see how this error can occur.

jhthomas, 2020-02-01:
Dear Jo,

All three functions now work on a LinksysOpenWRT1900AC even with the Base IP set to something other than 192.168.1.1 - 1.) Reboot to current partition 2.) Reboot to alternative partition and 3.) Perform Power Off.  The only caveat is that the fan keeps running continuously after the "power off".   My guess is that this is to be expected for this hardware.

Thank you.

Powered by LuCI openwrt-19.07 branch (git-20.031.59285-1592197) / OpenWrt 19.07.0 r10860-a3ffeb413b
luci-app-advanced-reboot	git-20.031.59285-1592197-51

commit 0a07d535 (2020-01-29): luci-app-advanced-reboot: properly handle multiple or missing lan IPs

Fixes: #3587
commit 19a48f64 (2020-01-29): luci-app-advanced-reboot: properly handle multiple or missing lan IPs

Fixes: #3587

(cherry picked from commit 0a07d53594b1b9152ea87e4be7366cc91ad36cc1)