openwrt/luci #3592: Question: Implementing XHR-Requests client-side context

dibdot, 2020-01-30:
Hi Jo,

just a quick one: I've started to migrate my apps from lua to client-side implementation. Is there a good (simple ;-)) example how to implement XHR-(Poll)-Requests in the new JS-framework?

Thanks!

jow-, 2020-01-30:
The code for that would be:

```javascript
L.Poll.add(function() {
    return L.Request.get(L.url('some/thing')).then(function(reply) {
        var data = reply.json();  // or whatever
        // do something with the data, e.g.
        var rows = ...;
        cbi_update_table('#my_status_table', rows);
    });
});
```

For further, more elaborate samples see the search: https://github.com/openwrt/luci/search?l=JavaScript&q=Poll.add

Maybe provide some specific code you'd like to convert, this would make it easier to answer your question.

dibdot, 2020-01-30:
Great, many thanks for your support!

closed 2020-01-30 by dibdot