openwrt/luci #3593: [request] add checkbox for installable or upgradable packages

userosos, 2020-01-31:
Hi! Can you add checkbox for upgradable packages and for installable packages. It will be good. I need install and upgrade only one packages now but i want install or upgrade all of them packages.

jow-, 2020-01-31:
It is not recommended to upgrade all packages at once since this will frequently lead to bricks or other undefined system behavior. Therefor, I'll not implement this.

closed 2020-01-31 by jow-

userosos, 2020-01-31:
> 
> 
> It is not recommended to upgrade all packages at once since this will frequently lead to bricks or other undefined system behavior. Therefor, I'll not implement this.

Ok. Can you add button for upgrade next package after successful upgrade one of package?