openwrt/luci #3605: Xiaomi router 3g Openwrt V19 problem. Upload very low.

roberdlv, 2020-02-05:
Hello, I have xiaomi router 3g, I installed OpenWrt v18 and everything perfect, perfect speed, perfect management (although improvable) etc ...

The problem comes when upgrading to v19 and v19.0.1, the router's upload speed low by half (wired 600mbs v18) v19 300mbs upload speed .... it looks like a bug and it happens to many people , is there any solution to this? Is it possible to improve the router driver and improve the upload and full management of v19? thanks.

feckert, 2020-02-05:
I would say this is not LuCI related please open a Ticket in https://bugs.openwrt.org/

closed 2020-02-05 by feckert

dangowrt, 2020-02-05:
For WiFi related issues on that specific device http://github.com/openwrt/mt76 would be the place to go.