openwrt/luci #3609: [19.07] luci-lib-ip: returns interface-identifier for v4-netmasks

SvenRoederer, 2020-02-06:
Based on this issue on our freifunk-firmware (https://github.com/freifunk-berlin/firmware/issues/754) I found the following since 1d2b4c777fd47f8d520b21d7085d3e7eb387cf02

(code taken from our wizard https://github.com/freifunk-berlin/firmware-packages/blob/1226ad42fe81eba6bd74df786320976cc154f006/utils/luci-app-ffwizard-berlin/luasrc/model/cbi/freifunk/assistent/wireless.lua#L236)

```lua
local ip = require "luci.ip"
local dhcpmeshnet = '10.230.197.209/28'

dhcpmeshnet = ip.IPv4(dhcpmeshnet)

local prenetconfig={}
prenetconfig.type='bridge'
prenetconfig.proto='static'
prenetconfig.ipaddr=dhcpmeshnet:minhost():string()
prenetconfig.netmask=dhcpmeshnet:mask():string()

print('prenet.addr: ' .. prenetconfig.ipaddr)
print('prenet.mask: ' .. prenetconfig.netmask)
print('dhcpmesh.mask: ' .. dhcpmeshnet:minhost():string())
```
Running this code results in:
> prenet.addr: 10.230.197.209
> prenet.mask: 255.255.255.240%ifb0
> dhcpmesh.mask: 255.255.255.240

Commenting out the line `prenetconfig.proto='static'` changes the output to:
> prenet.addr: 10.230.197.209
> prenet.mask: 255.255.255.240
> dhcpmesh.mask: 255.255.255.240

We use the prenetconfig variable to generate a new interface in uci, which ends up in an invalid network-config as of the faulty netmask. This appended interface we have never seen before the change of lib-ip. Even commit b349428e167c5c129ce352aa6d391fea85122fd4 doesn't help.

@jow- any ideas?

SvenRoederer, 2020-02-06:
This can also be seen on master branch.

jow-, 2020-02-07:
I guess there's more places where the scope member is not properly initialized to 0. If you have a reliable reproducer, try debugging it.

SvenRoederer, 2020-02-07:
Just downloaded a fresh snapshot for x86-generic from the buildbot and ran the following:

```
BusyBox v1.31.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r12202-cc89c5fe27
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:~# opkg update
root@OpenWrt:~# opkg install lua luci-lib-ip
root@OpenWrt:/tmp# opkg list luci-lib-ip
luci-lib-ip - git-20.038.16141-980e893-1 - Lua library for IP calculation and routing information
root@OpenWrt:~# cd /tmp/
root@OpenWrt:/tmp# cat test-libip.lua 
local ip = require "luci.ip"
local dhcpmeshnet = '10.230.197.209/28'

dhcpmeshnet = ip.IPv4(dhcpmeshnet)

local prenetconfig={}
prenetconfig.type='bridge'
prenetconfig.proto='static'
prenetconfig.ipaddr=dhcpmeshnet:minhost():string()
prenetconfig.netmask=dhcpmeshnet:mask():string()

print('prenet.addr: ' .. prenetconfig.ipaddr)
print('prenet.mask: ' .. prenetconfig.netmask)
print('dhcpmesh.mask: ' .. dhcpmeshnet:minhost():string())

root@OpenWrt:/tmp# lua test-libip.lua 
prenet.addr: 10.230.197.209
prenet.mask: 255.255.255.240%br-lan
dhcpmesh.mask: 10.230.197.209
root@OpenWrt:/tmp#
```
That's exact the code provided above causing this problem.
@jow- you don't see this faulty result? Then the question comes up, what's the difference in our setups?

Can anyone else reproduce?

jow-, 2020-02-07:
This happens on 19.07.1 x86/64 in Qemu:

```
BusyBox v1.30.1 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 19.07.1, r10911-c155900f66
 -----------------------------------------------------
=== WARNING! =====================================
There is no root password defined on this device!
Use the "passwd" command to set up a new password
in order to prevent unauthorized SSH logins.
--------------------------------------------------
root@OpenWrt:/# vi /tmp/test.lua
root@OpenWrt:/# cat /tmp/test.lua 
local ip = require "luci.ip"
local dhcpmeshnet = '10.230.197.209/28'

dhcpmeshnet = ip.IPv4(dhcpmeshnet)

local prenetconfig={}
prenetconfig.type='bridge'
prenetconfig.proto='static'
prenetconfig.ipaddr=dhcpmeshnet:minhost():string()
prenetconfig.netmask=dhcpmeshnet:mask():string()

print('prenet.addr: ' .. prenetconfig.ipaddr)
print('prenet.mask: ' .. prenetconfig.netmask)
print('dhcpmesh.mask: ' .. dhcpmeshnet:minhost():string())
root@OpenWrt:/# lua /tmp/test.lua
prenet.addr: 10.230.197.209
prenet.mask: 255.255.255.240
dhcpmesh.mask: 10.230.197.209
root@OpenWrt:/# 
```

closed 2020-02-08 by jow-

SvenRoederer, 2020-02-14:
Thanks for fixing and for cherry-picking to 19.07.

Even I still wonder, why removing the ```prenetconfig.proto='static'``` line changes the result copied from lib-ip ...

commit c2eeb09b (2020-02-08): luci-lib-ip: explicitly clear scope when allocating new userdata

Fixes: #3609
commit 8b168352 (2020-02-08): luci-lib-ip: explicitly clear scope when allocating new userdata

Fixes: #3609

(cherry picked from commit c2eeb09bd99b03eead837d8dbfb3f0d8046ccc65)