openwrt/luci #3612: Security bug : Arbitrary File Creation

delikely, 2020-02-07:
There is no restriction on the path of the system log file(`logread`),which lead to **overwritten important file**,such as /etc/passwd. if `/etc/passwd` was overwritened,it will cause a denial of service. when a router was infected, The only way to fix  is **Flash Firmware**.

I already reported it at [bugs.openwrt.org](https://bugs.openwrt.org/index.php?do=details&task_id=2739&opened=2412&openedsm=userid) a faw days,but nobody to deal with.

ysc3839, 2020-02-07:
But modifying config requires login as root. With root permission you can modify any file without restriction.

jow-, 2020-02-07:
Exactly. You could also add crontab entries overwriting arbitrary files, local startup commands deleting the root filesystem etc.

closed 2020-02-07 by jow-

delikely, 2020-02-07:
@ysc3839  @jow-  we can trigger this throght web page.  [poc]( https://github.com/delikely/Arbitrary-File-Creation-in-OpenWrt)

jow-, 2020-02-07:
@delikely - yes, I can also it trigger it through the webpage by putting `rm -f /etc/passwd` into System -> Startup -> Local Startup. I could also disable uhttpd and dropbear in System -> Startup to achive a similar effect. There is no way we can handle all possible corner cases in the UI.